miskcoo/ugreen_leds_controller #6: Is it possible to use the control scripts on the DXP480T?

blaugrau90, 2024-06-09:
I have the DXP480T Full SSD NAS. Unfortunately, the scripts do not work here. The SMBUS I801 is present, but the script does not find any LEDs that it can control.

Here is the output of i2cdetect -l and i2cdetect -y 0:

```~# i2cdetect -l
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	i2c       	i915 gmbus dpa                  	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-5	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-6	i2c       	i915 gmbus tc1                  	I2C adapter
i2c-7	i2c       	i915 gmbus tc2                  	I2C adapter
i2c-8	i2c       	i915 gmbus tc3                  	I2C adapter
i2c-9	i2c       	i915 gmbus tc4                  	I2C adapter
i2c-10	i2c       	i915 gmbus tc5                  	I2C adapter
i2c-11	i2c       	i915 gmbus tc6                  	I2C adapter
i2c-12	i2c       	AUX B/DDI B/PHY B               	I2C adapter
i2c-13	i2c       	AUX USBC1/DDI TC1/PHY TC1       	I2C adapter
i2c-14	i2c       	AUX USBC2/DDI TC2/PHY TC2       	I2C adapter
i2c-15	i2c       	AUX USBC3/DDI TC3/PHY TC3       	I2C adapter
i2c-16	i2c       	AUX USBC4/DDI TC4/PHY TC4       	I2C adapter

~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- 26 -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- 44 -- -- -- 48 -- 4a -- -- -- -- --
50: 50 -- 52 -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Is there any way to adapt this so that it also works on this NAS? Currently the power LED flashes all the time. There are no other LEDs.

miskcoo, 2024-06-09:
I think it is possible to control the power LED, but we need to probe each of these addresses.

According to what I see from the driver, you can read bytes from registers at `0x5a`, `0x5b` and `0x5d` using:
```bash
# replace <chip addr> below with the addresses you see from i2cdetect
i2cget -y 1  <chip addr>  0x5a b
i2cget -y 1  <chip addr>  0x5b b
sleep 5
i2cget -y 1  <chip addr>  0x5d b
```
If I am correct, the chip for the power LED will return `0xa5`, `0xb5` and the version of this chip, respectively. If you find there is an address returning these outputs, I can check how to turn on/off this LED later.

blaugrau90, 2024-06-09:
This is the result of the few commands for my chip address:

```
#Chip Address
addresses=(0x08 0x26 0x44 0x48 0x4a 0x50 0x52)

for addr in "${addresses[@]}"; do
  i2cget -y 0 $addr 0x5a b
  i2cget -y 0 $addr 0x5b b
done

sleep 5

for addr in "${addresses[@]}"; do
  i2cget -y 0 $addr 0x5d b
done

# Result from the first two commands
Error: Read failed
Error: Read failed
0xa5
0xb5
0x00
0x00
0x00
0x00
0x00
0x00
0x00
0x00
0x00
0x00

# Sleep Time 5 sec

# Result of the last command
Error: Read failed
0x06
0x00
0x00
0x00
0x00
0x00
```

miskcoo, 2024-06-09:
Good to see that, it seems `0x26` is the correct address. 

It seems the relevant registers at chip `0x26` are `0x50, 0x51, 0xb1, 0xa0, 0xe0, 0xe1`. All of them are 8-bits, so we need to use `i2cset -y 0 0x26 <reg addr> <data (a byte)> b`.

- The LED is initialized by sequentially writing:
```bash
i2cset -y 0 0x26 0x50 0 b
i2cset -y 0 0x26 0x51 0 b
i2cset -y 0 0x26 0xb1 2 b
i2cset -y 0 0x26 0xa0 1 b
```
- `0xb1` appears a control register, a typical value to write is of the form `offset << mode`, where `offset = 1, 2, 3` and I guess `mode = 0, 1` but not sure. You can try to write something to this register to see if the light will turn off. I seems writing something with (offset = 3, i.e., write 3 or 6 or ...) will turn it off.
- For setting the brightness, the code looks like there are two different command sequences:
  1. Write something (offset = 2) to `0xb1` and return;  
  2. Write something (offset = 1) to `0xb1`, and then write something like `offset2 << mode2` to `0xa0`, and finally write the same byte to both `0xe0, 0xe1`, indicating a brightness.
- For setting the color, the driver write to one of  `0x50` or `0x51` (according a variable, not sure its meaning); and then write to `0xb1` with offset = 3; and then write something to `0xa0`. 

I guess you can search these undermined values and try to figure the meaning of them.

miskcoo, 2024-06-09:
For reading the status, you can try registers starting from `0x80` (maybe `0x80` and `0x81`, the status is represented by 2 bytes). They are also read in byte mode.

miskcoo, 2024-06-09:
For blinking and breathing, one pattern is:
1. writing something to `0x50`; 
2. then write something with mode = 1 to `0xb1`; 
3. then write something to `0x51`;

Finally, for consecutive writes, you may need to add a sleep between them.

blaugrau90, 2024-06-09:
With the following code, I was able to stop the LED from flashing. That was very annoying. I assume that the color cannot be changed, it seems to be a white LED. Not an RGB LED.

```
i2cset -y 0 0x26 0x50 0 b
i2cset -y 0 0x26 0x51 0 b
i2cset -y 0 0x26 0xb1 2 b
i2cset -y 0 0x26 0xa0 1 b
```

Even though I haven't quite understood what each line of it does. I think `i2cset -y 0 0x26 0xb1 2 b` has no influence on the LED. At least it lights up continuously even without the command.
Could you explain the individual commands to me in detail and what effect they have on the LED?

miskcoo, 2024-06-09:
I do not know the exact meaning of them, neither. It was found in the de-compiled code of the kernel module named `leds-mcu-n76e003.ko`. To determine their meaning, we need to either look at the code in detail, but it requires many efforts; or alternatively, with a device at hand (I don't have it, so I cannot do this), we can try to write I2C registers according to the observed command sequences in that module, and check the LED's status.

Basically, I think `i2cset -y 0 0x26 0xb1 <data> b` is used to change the basic state (on / off / blink) of this LED; and similarly for `i2cset -y 0 0x26 0xa0 <data> b`; and `i2cset -y 0 0x26 0x50 <data> b` and `i2cset -y 0 0x26 0x51 <data> b` look like the additional information (e.g., the blink delay; color?). 

I guess to turn-off the light, you can use `i2cset -y 0 0x26 0xb1 3 b`. To investigate further, you can try to run a grid search and observe the status of your LED. For example, 

```bash
# change only one register
for i in $(seq 0 7); do
    i2cset -y 0 0x26 0xb1 $i b
    sleep 1
done

# change two registers
for i in $(seq 0 7); do
    for j in $(seq 0 7); do
        i2cset -y 0 0x26 0xb1 $i b
        sleep 0.5
        i2cset -y 0 0x26 0xa0 $j b
    done
    sleep 1
done
````

blaugrau90, 2024-06-09:
I have found out which commands are suitable for what. 
Pulsing: 
Permanently on
i2cset -y 0 0x26 0x50 0 b

Fast flashing
i2cset -y 0 0x26 0x50 1 b

Slow flashing
i2cset -y 0 0x26 0x51 1 b

Breathing slowly
i2cset -y 0 0x26 0x50 2 b

Colors: 
Red LED on after off
`i2cset -y 0 0x26 0xb1 1 b` or `i2cset -y 0 0x26 0xb1 5 b` 
Red LED off after on
`i2cset -y 0 0x26 0xa0 1 b` or `i2cset -y 0 0x26 0xa0 5 b`

White LED on after off
`i2cset -y 0 0x26 0xb1 2 b` or `i2cset -y 0 0x26 0xb1 6 b`
White LED off after on
`i2cset -y 0 0x26 0xa0 2 b` or `i2cset -y 0 0x26 0xa0 6 b`

If both LEDs are on, the one that was switched on first wins.
If this is then switched off, the other LED that is still switched on remains on.
After the LED is switched from Breathering to permanent, it remains at the current brightness level. depending on when you switch, it may be almost completely dark or completely bright. You then have to switch it back to full brightness with the 0xb1.

blaugrau90, 2024-06-09:
Summarized here, every possible LED function that seems to exist. 

```
### Applies to the UGREEN DXP480T NAS ###

## Permanently ON
## i2cset -y 0 0x26 0x50 0 b

## Fast Flashing
## i2cset -y 0 0x26 0x50 1 b

## Breathing
## i2cset -y 0 0x26 0x50 2 b

## Slowly Flashing
## i2cset -y 0 0x26 0x51 1 b

## Preset for Effects:
## i2cset -y 0 0x26 0xa0 1 b # red off
## i2cset -y 0 0x26 0xa0 2 b # white off
## i2cset -y 0 0x26 0x50 0 b # Permanently on

## red on
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off

i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 0 b # Permanently on

## white on
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off

i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 0 b # Permanently on

## red Fast Flashing
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x50 0 b # Permanently on

i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 1 b # Fast Flash on

## white Fast Flashing
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x50 0 b # Permanently on

i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 1 b # Fast Flash on

## red Slowly Flashing
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x50 0 b # Permanently on

i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x50 0 b # Fast Flash / Breathing off
i2cset -y 0 0x26 0x51 1 b # Slow Flash on

## white Slowly Flashing
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x50 0 b # Permanently on

i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x50 0 b # Fast Flash / Breathing off
i2cset -y 0 0x26 0x51 1 b # Slow Flash on

## red Breathing
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x50 0 b # Permanently on

i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 2 b # Breathing on

## white Breathing
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x50 0 b # Permanently on

i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 2 b # Breathing on
```

Erani0, 2024-06-10:
Oh nice I just knew it was on the HDD Nas systems. What would that look like if you wanted to use it? I've never worked with anything like this before.

blaugrau90, 2024-06-10:
My primary concern was that the power LED kept flashing after I replaced the UGOS with a different OS. That's why I wanted to know how to control the LED. I found out with the help of this repo and the developer. Now a script always runs when the system boots, which sets the LED to permanently white.

Erani0, 2024-06-10:
I use Proxmox now

An i have no idea ro write Linux scripts

YugiFanGX, 2024-06-14:
TrueNAS Scale, no output from i2cdetect -l ...

miskcoo, 2024-06-14:
do you load the i2c-dev module? please check the output of lsmod.

YugiFanGX, 2024-06-14:
That solved the issue. Thank you!

Dataninja126, 2024-06-21:
I'm passing the SATA controllers for Proxmox into a TrueNAS VM and I believe the passing of the controllers may be affecting the device IDs?  None of the commands in the exhaustive list above work and I get an `Error: Write failed` in both.  Any suggestions?

Here is the output of i2cdetect -l and i2cdetect -y 0 in TrueNAS VM:
`i2c-0   smbus           SMBus PIIX4 adapter at 0700             SMBus adapter

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 51 52 53 54 55 56 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
`
Here is the output of i2cdetect -l and i2cdetect -y 0 in Proxmox:

`i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-3   i2c             i915 gmbus dpa                          I2C adapter
i2c-4   i2c             i915 gmbus dpb                          I2C adapter
i2c-5   i2c             i915 gmbus dpc                          I2C adapter
i2c-6   i2c             i915 gmbus tc1                          I2C adapter
i2c-7   i2c             i915 gmbus tc2                          I2C adapter
i2c-8   i2c             i915 gmbus tc3                          I2C adapter
i2c-9   i2c             i915 gmbus tc4                          I2C adapter
i2c-10  i2c             i915 gmbus tc5                          I2C adapter
i2c-11  i2c             i915 gmbus tc6                          I2C adapter
i2c-12  i2c             AUX B/DDI B/PHY B                       I2C adapter
i2c-13  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
i2c-14  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter
i2c-15  i2c             AUX USBC3/DDI TC3/PHY TC3               I2C adapter
i2c-16  i2c             AUX USBC4/DDI TC4/PHY TC4               I2C adapter

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- 3a -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- 48 -- 4a -- -- -- -- -- 
50: 50 -- 52 -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  `

miskcoo, 2024-06-21:
@Dataninja126 at the moment you cannot use this tool in a VM -- it should be run in the host (i.e., your proxmox);

for the outputs in proxmox, is `i2c-0` missing? and what's your NAS model?

Dataninja126, 2024-06-21:
> @Dataninja126 at the moment you cannot use this tool in a VM -- it should be run in the host (i.e., your proxmox);
> 
> for the outputs in proxmox, is `i2c-0` missing? and what's your NAS model?

Sorry I must have accidently omitted that line

```i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-3   i2c             i915 gmbus dpa                          I2C adapter
i2c-4   i2c             i915 gmbus dpb                          I2C adapter
i2c-5   i2c             i915 gmbus dpc                          I2C adapter
i2c-6   i2c             i915 gmbus tc1                          I2C adapter
i2c-7   i2c             i915 gmbus tc2                          I2C adapter
i2c-8   i2c             i915 gmbus tc3                          I2C adapter
i2c-9   i2c             i915 gmbus tc4                          I2C adapter
i2c-10  i2c             i915 gmbus tc5                          I2C adapter
i2c-11  i2c             i915 gmbus tc6                          I2C adapter
i2c-12  i2c             AUX B/DDI B/PHY B                       I2C adapter
i2c-13  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
i2c-14  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter
i2c-15  i2c             AUX USBC3/DDI TC3/PHY TC3               I2C adapter
i2c-16  i2c             AUX USBC4/DDI TC4/PHY TC4               I2C adapter```

miskcoo, 2024-06-21:
> > @Dataninja126 at the moment you cannot use this tool in a VM -- it should be run in the host (i.e., your proxmox);
> > for the outputs in proxmox, is `i2c-0` missing? and what's your NAS model?
> 
> Sorry I must have accidently omitted that line and I have the 6800Pro
> 
> ```
> i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
> i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
> i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
> i2c-3   i2c             i915 gmbus dpa                          I2C adapter
> i2c-4   i2c             i915 gmbus dpb                          I2C adapter
> i2c-5   i2c             i915 gmbus dpc                          I2C adapter
> i2c-6   i2c             i915 gmbus tc1                          I2C adapter
> i2c-7   i2c             i915 gmbus tc2                          I2C adapter
> i2c-8   i2c             i915 gmbus tc3                          I2C adapter
> i2c-9   i2c             i915 gmbus tc4                          I2C adapter
> i2c-10  i2c             i915 gmbus tc5                          I2C adapter
> i2c-11  i2c             i915 gmbus tc6                          I2C adapter
> i2c-12  i2c             AUX B/DDI B/PHY B                       I2C adapter
> i2c-13  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
> i2c-14  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter
> i2c-15  i2c             AUX USBC3/DDI TC3/PHY TC3               I2C adapter
> i2c-16  i2c             AUX USBC4/DDI TC4/PHY TC4               I2C adapter
> ```

Have you tried the cli tool in proxmox? I think it will work

Dataninja126, 2024-06-21:
> > > @Dataninja126 at the moment you cannot use this tool in a VM -- it should be run in the host (i.e., your proxmox);
> > > for the outputs in proxmox, is `i2c-0` missing? and what's your NAS model?
> > 
> > 
> > Sorry I must have accidently omitted that line and I have the 6800Pro
> > ```
> > i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
> > i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
> > i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
> > i2c-3   i2c             i915 gmbus dpa                          I2C adapter
> > i2c-4   i2c             i915 gmbus dpb                          I2C adapter
> > i2c-5   i2c             i915 gmbus dpc                          I2C adapter
> > i2c-6   i2c             i915 gmbus tc1                          I2C adapter
> > i2c-7   i2c             i915 gmbus tc2                          I2C adapter
> > i2c-8   i2c             i915 gmbus tc3                          I2C adapter
> > i2c-9   i2c             i915 gmbus tc4                          I2C adapter
> > i2c-10  i2c             i915 gmbus tc5                          I2C adapter
> > i2c-11  i2c             i915 gmbus tc6                          I2C adapter
> > i2c-12  i2c             AUX B/DDI B/PHY B                       I2C adapter
> > i2c-13  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
> > i2c-14  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter
> > i2c-15  i2c             AUX USBC3/DDI TC3/PHY TC3               I2C adapter
> > i2c-16  i2c             AUX USBC4/DDI TC4/PHY TC4               I2C adapter
> > ```
> 
> Have you tried the cli tool in proxmox? I think it will work. This issue and the commands above are about 480T -- it is different from other HDD models.

Well, the script worked in a sense that it stopped the trailing lights, but because the SATA controllers are passed through it's triggering the following:

```
Mapping devices...
Hardware mapping (hwmap):
Checking zpool status...
no pools available
Processing  with status 
./ugreen_leds6800.sh: line 49: hwmap: bad array subscript
Warning: No mapping found for 
Final device statuses:
0: p
1: u
2: x
3: x
4: x
5: x
6: x
7: x
```

miskcoo, 2024-06-21:
> Well, the script worked in a sense that it stopped the trailing lights, but because the SATA controllers are passed through it's triggering the following:
>
> ```
> Mapping devices...
> Hardware mapping [...]

I think this is an expected behavior, supporting this case will be complicated.

Dataninja126, 2024-06-21:
> I think this is an expected behavior, supporting this case will be complicated, though it is possible.

Thanks for the quick replies. I'm okay with the lights not flashing.

Because I'm not well versed in this space, I was considering passing through i2c-0, but not sure how that would affect my Proxmox system...

```
i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
```

bangity, 2024-06-30:
> My primary concern was that the power LED kept flashing after I replaced the UGOS with a different OS. That's why I wanted to know how to control the LED. I found out with the help of this repo and [...]

@blaugrau90 can you please share this script? I have the same machine and tried Proxmox, not sure why it kept flashing until now. I also get random reboot and intermittenly lose Web GUI Proxmox activity. Have you experience that?

Thanks.

MrGKanev, 2024-07-22:
Is there a way to turn them off?

GO1984, 2024-09-26:
I have installed Windows Server 2025 on my DXP480T after some hardware upgrades.
The blinking Power-LED is very annoying. Is there any Love for a Windows OS too?

datarepos76, 2024-10-20:
> @blaugrau90 can you please share this script? I have the same machine and tried Proxmox, not sure why it kept flashing until now. I also get random reboot and intermittenly lose Web GUI Proxmox [...]

@blaugrau90 I agree with the previous speaker. It would be great if you could make your script available here. Many thanks in advance!

blaugrau90, 2024-10-20:
@bangity 
I have not yet noticed any unexpected reboots of my Proxmox installation. This probably has nothing to do with the control of the LED. 

@miskcoo Maybe you can add what comes next to your repo / wiki.

@bangity @datarepos76 

I'll share how I implemented this. 

First, we need two scripts that control the LED after booting and when shutting down.
When booting, my LED flashes white and then lights up permanently. 
When I shut down the system, it flashes red and then switches off when the system is off.

We then create startup and shutdown services that use these scripts.

Here is a detailed description of how this works.

To ensure a script runs during boot and another during shutdown on Debian, you can use the init system settings. Depending on the version of Debian, you can use `systemd`, which is the default init system. Here's a guide on how to do this:

### startup-script.sh

```bash
#!/bin/bash
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 0 b # Permanently on
i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x50 1 b # Fast Flash on

sleep 10

i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x50 0 b # Permanently on
sleep 1
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0xb1 2 b # white on
i2cset -y 0 0x26 0x50 0 b # Permanently on
```

### shutdown-script.sh

```bash
#!/bin/bash
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0x50 0 b # Permanently on
i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x50 1 b # Fast Flash on

sleep 5

i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x50 0 b # Permanently on
sleep 1
i2cset -y 0 0x26 0xa0 1 b # red off
i2cset -y 0 0x26 0xa0 2 b # white off
i2cset -y 0 0x26 0x51 0 b # Slow Flash off
i2cset -y 0 0x26 0xb1 1 b # red on
i2cset -y 0 0x26 0x50 0 b # Permanently on
```

### Running a script at boot

1. **Create the script**: Create a script that should be executed at boot. Example:
    ```bash
    sudo nano /usr/local/bin/startup-script.sh
    ```

2. **Add content to the script**:
    ```bash
    #!/bin/bash
    echo "System is starting up" >> /var/log/startup-script.log
    # Add your commands here
    ```

3. **Make the script executable**:
    ```bash
    sudo chmod +x /usr/local/bin/startup-script.sh
    ```

4. **Create a systemd service**:
    ```bash
    sudo nano /etc/systemd/system/startup-script.service
    ```

5. **Configure the service file**:
    ```ini
    [Unit]
    Description=Startup Script

    [Service]
    Type=oneshot
    ExecStart=/usr/local/bin/startup-script.sh

    [Install]
    WantedBy=multi-user.target
    ```

6. **Enable the service**:
    ```bash
    sudo systemctl enable startup-script.service
    ```

### Running a script at shutdown

1. **Create the script**: Create a script that should be executed at shutdown. Example:
    ```bash
    sudo nano /usr/local/bin/shutdown-script.sh
    ```

2. **Add content to the script**:
    ```bash
    #!/bin/bash
    echo "System is shutting down" >> /var/log/shutdown-script.log
    # Add your commands here
    ```

3. **Make the script executable**:
    ```bash
    sudo chmod +x /usr/local/bin/shutdown-script.sh
    ```

4. **Create a systemd service**:
    ```bash
    sudo nano /etc/systemd/system/shutdown-script.service
    ```

5. **Configure the service file**:
    ```ini
    [Unit]
    Description=Shutdown Script
    DefaultDependencies=no
    Before=shutdown.target reboot.target halt.target

    [Service]
    Type=oneshot
    ExecStart=/usr/local/bin/shutdown-script.sh

    [Install]
    WantedBy=halt.target reboot.target shutdown.target
    ```

6. **Enable the service**:
    ```bash
    sudo systemctl enable shutdown-script.service
    ```

### Summary

1. Create the scripts and make them executable.
2. Create and configure the respective `systemd` service files.
3. Enable the services so that they are executed during boot and shutdown.

By following these steps, your startup script will run at system boot, and your shutdown script will run when the system is shutting down.

mazo6515, 2025-06-29:
Hello I run it also on DXP480T - windows server 2022 and tried it under windows ubuntu subsystem - the i2cdetect -l  command gives no response or feedback   and the     i2cdetect -y O:      gives Error: I2C bus name doesn´t match any bus present!   - and usage cases    - is there any way to get it run in WS2022 environment?   (I have installed and implemented step by step the startup and shutdown script as above...

blaugrau90, 2026-03-17:
It should work. As far as I know, Truenas Scale is based on Debian. The I2C tools should be available there, or at least they can be installed later. Truenas Core is based on FreeBSD. The I2C tools are different there. The script might work, but it might not. I haven’t tested it.

yuryroot, 2026-09-30:
> So this also work on TrueNAS right?

@dffvb @blaugrau90 on TrueNAS I had to add:
```
modprobe -v i2c-dev
```
before sets of `i2cset` commands in both startup and shutdown scripts.

Without it I was getting such errors:
```
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
```

dffvb, 2026-09-30:
@yuryroot will try, many thanks!

datarepos76, 2026-10-15:
Hi,

I have this starup script installed on my DXP-480T Plus. It seems not to work, the LED remains in blinking status. The Plugin is installed.

#!/bin/bash
I2C_DEV=$(i2cdetect -l | grep "SMBus I801 adapter" | grep -Po "i2c-\d+" | cut -d'-' -f2)
modprobe -v i2c-dev
i2cset -y ${I2C_DEV} 0x26 0xa0 1 b  # red off
i2cset -y ${I2C_DEV} 0x26 0xa0 2 b  # white off
i2cset -y ${I2C_DEV} 0x26 0xb1 2 b  # white on
i2cset -y ${I2C_DEV} 0x26 0x51 0 b  # slow flash off
i2cset -y ${I2C_DEV} 0x26 0x50 0 b  # permanently on

Do you have any ideas?

Thx and best

datarepos

[image]