miskcoo/ugreen_leds_controller #7: How to configure service when using dkms?

manutenfruits, 2024-06-12:
I installed the .deb package in debian, and it seems to work. I get all the devices under `/sys/class/leds` and I can manipulate them with `echo` commands.

However, I want to change the device in `/sys/class/leds/netdev/device_name`, as it doesn't match my interface name. If I use `echo`, it will be reset after next reboot.

Additionally, the disk LEDs don't appear to work out of the box. I am writing using `dd` to `/dev/sda` and the led stays solid white. I couldn't find a way to check what device `/sys/class/leds/disk1` should follow, but `lsmod` shows me that both `ledtrig_netdev` and `ledtrig_oneshot` are loaded.

Any pointers? Thanks for building this!

PS: DXP6800 Pro (6 bay) is also supported by this :)

miskcoo, 2024-06-12:
The scripts in the released package is outdated now. You can copy current `scripts/ugreen-diskiomon`, `scripts/ugreen-netdevmon` and `scripts/ugreen-probe-leds` to `/usr/bin`, and create a file `/etc/systemd/system/ugreen-ledmon@.service` with the following content:
```
[Unit]
Description=UGREEN LEDs daemon for monitoring diskio / netio (of %i) and blinking corresponding LEDs

[Service]
ExecStartPre=/usr/bin/ugreen-probe-leds
ExecStartPre=/usr/bin/ugreen-netdevmon %i
ExecStart=/usr/bin/ugreen-diskiomon
StandardOutput=journal

[Install]
WantedBy=multi-user.target
```

Then, you can use `systemctl start ugreen-ledmon@enp2s0` to specify which interface you want to monitor.

The disk activities are monitored by `ugreen-diskiomon`, and the map of devices and LEDs is also defined in this file. It uses the HCTL addresses by default (we do not know whether it will change, but it is stable for my device, see the discussions in #4 ). If you know the serial number of disks, you can change the `serial_map=("placeholder0", ...` array in that file, and `mapping_method=hctl` to `mapping_method=serial`.

I may create a new release and update the doc later.

meyergru, 2024-06-12:
Apart from that, at this time, we cannot Monitor disk activity, only ZFS Status or smart Status. This is different for Networking.

miskcoo, 2024-06-12:
> Apart from that, at this time, we cannot Monitor disk activity, only ZFS Status or smart Status. This is different for Networking.

Although there are no triggers for it, I think we can blink LEDs based on the change of `/sys/class/block/*/stat`?

meyergru, 2024-06-12:
I meant: At this Time ;-) My comment was directed at the obvious expectation to have that Feature by people who cannot read code. The Facebook Forum is Full of people who cannot Even use this repo…

closed 2024-06-15 by miskcoo