miskcoo/ugreen_leds_controller #8: Is it possible to control LEDs in DSM?

kernel-t4sk, 2024-06-13:
My DX4600Pro is running DSM-7.2.1 (SA6400). I'm trying to compile it for use in DSM, but unfortunately, DSM seems to lack tools for i2c-dev. Is there any other approach I can try? many thanks.

Lotusice, 2024-06-16:
Maybe I2C-I801 modules are required. This is a tutorial that can control six LEDs on DX4600Pro (in Chinese).
https://www.bilibili.com/read/cv35337126/

closed 2024-06-18 by kernel-t4sk

TheUntouchable, 2024-06-30:
@kernel-t4sk  Did you get the LEDs working on DSM?

h0we, 2024-07-02:
> @kernel-t4sk Did you get the LEDs working on DSM?

yes, it can operate, but it can't do real-time monitoring of the LAN and disk statuses. The LEDs can only stay constantly lit and can't blink based on usage status.

Chris-V702, 2024-07-24:
Someone got it working on the DXP8800 Plus also with DSM (SA6400) and can share his how to?

TheUntouchable, 2024-07-24:
> Someone got it working on the DXP8800 Plus also with DSM (SA6400) and can share his how to?

When you are using ARC Loader there is already a plug-in for controlling the LEDs, problem is that the job to update the LEDs did take so much CPU that this has currently been deactivated