miskcoo/ugreen_leds_controller #13: TrueNAS Scale

sgreiner, 2024-06-20:
The following instructions are a memory log from my successful attempt today to get the LEDs of the Ugreen NASync DXP6800 Pro running poperly on TrueNAS Scale.

I hope I haven't forgotten anything, otherwise I'll add to it when I reinstall it (probably next weekend). Big thanks to @miskcoo for your work on this repository!

I thought someone might find this useful, so I add this as issue here, hope thats ok. There is also room for improvement to add `zpool status` related output to the leds (e.g. make them red if a pool is degraded etc).

# Ugreen NAS LEDs on TrueNAS Scale
20.06.2024

## Builder VM
Install TrueNAS Scale on a VM. These steps could be done on the target NAS, but we don't want to clutter the appliance more than necessary.

You might need to adjust the version numbers below in the future.

Execute as root (`sudo su -`) on a fresh TrueNAS VM with the exact same version as your target TrueNAS Scale.

```bash
export PATH=/usr/bin:/usr/sbin

mount -o remount,rw 'boot-pool/ROOT/24.04.1.1'
mount -o remount,rw 'boot-pool/ROOT/24.04.1.1/usr'

chmod +x /bin/apt*
chmod +x /usr/bin/dpkg

apt update

apt install build-essential linux-headers-generic

cd /root
git clone https://github.com/miskcoo/ugreen_dx4600_leds_controller.git

cd /root/ugreen_dx4600_leds_controller/cli && make
cd /root/ugreen_dx4600_leds_controller/kmod && make
```

copy these two files to your target TrueNAS somehow:

```bash
# example
scp /root/ugreen_dx4600_leds_controller/cli/ugreen_leds_cli root@target:/root
scp /root/ugreen_dx4600_leds_controller/kmod/led-ugreen.ko root@target:/root
```

## Target TrueNAS (e.g. on a Ugreen NASync DXP6800 Pro)

install TrueNAS Scale:
- insert two NVMEs into the NAS as boot-pool mirror
- enter BIOS (CTRL+F12)
- disable the watchdog in the BIOS
- disable the NVME slot with the Ugreen OS on it in the BIOS 
- install TrueNAS Scale from a USB stick (flash the ISO with Balena Etcher onto the USB stick)

On TrueNAS Scale 24.04.1.1 dependencies like i2c-dev, ledtrig-oneshot, ledtrig-netdev and smartmontools are available.

execute as root on your target TrueNAS:

```bash
cd /root
git clone https://github.com/miskcoo/ugreen_dx4600_leds_controller.git

# until next reboot we need write access
mount -o remount,rw 'boot-pool/ROOT/24.04.1.1/usr'
mount -o remount,rw 'boot-pool/ROOT/24.04.1.1/etc'
```

Create kernel modules load config:
```bash
vim /etc/modules-load.d/ugreen-led.conf
```
content:
```
i2c-dev
led-ugreen
ledtrig-oneshot
ledtrig-netdev
```

Put previously compiled kernel module in place (kernel version might change)
```bash
cp led-ugreen.ko /lib/modules/6.6.29-production+truenas/extra
```

Run depmod and modprobe:
```bash
depmod
modprobe -a i2c-dev led-ugreen ledtrig-oneshot ledtrig-netdev
```

systemd setup for leds
```bash
cd /root/ugreen_dx4600_leds_controller/
```

modify the scripts now if you want other colors etc
```bash
scripts=(ugreen-diskiomon ugreen-netdevmon ugreen-probe-leds)
for f in ${scripts[@]}; do
    chmod +x "scripts/$f"
    cp "scripts/$f" /usr/bin
done

cp scripts/*.service /etc/systemd/system/

systemctl daemon-reload

# change name of NIC if needed
systemctl start ugreen-ledmon@enp88s0
systemctl enable ugreen-ledmon@enp88s0
```

Reboot and verify.

After a TrueNAS Scale Update you need to repeat all steps, as the boot-pool datasets are versioned and our changes will be gone in a new boot-pool dataset.

miskcoo, 2024-06-20:
The instruction looks good, and don't forget to install `dmidecode` (especially for 6800 Pro). It will be used in the script to detect the device type and to choose the correct htcl mapping.

For ZFS, I am also using it. I am curious about how the zpool is configured on TrueNAS. Could `zpool status -L | egrep "^\s*(sd|dm)"` give the correct devices?

sgreiner, 2024-06-20:
> The instruction looks good, and don't forget to install `dmidecode` (especially for 6800 Pro). It will be used in the script to detect the device type and to choose the correct htcl mapping.

`dmidecode` is also available per default (version 3.4)

I will check the zpool stuff on the ata-disk-mapping branch

miskcoo, 2024-06-21:
Not sure if this will work. 

Maybe we can skip installing a VM by building the kmod in a Debian docker with the headers from truenas? I can build such a module using the scripts in the `truenas-build` branch.

Adrian-Grimm, 2024-07-24:
@sgreiner may possible to run all this in a Debian Jail with required capabilities of https://github.com/Jip-Hop/jailmaker ? 
With that it would survive a TrueNAS update...

sgreiner, 2024-07-30:
> Not sure if this will work.
> 
> Maybe we can skip installing a VM by building the kmod in a Debian docker with the headers from truenas? I can build such a module using the scripts in the `truenas-build` branch.

Today I updated my TrueNAS. I cloned this repo to my Desktop PC (with wsl2 and docker desktop) and run your truenas build script, copied the kernel module to the new TrueNAS (24.04.2) it worked flawlessly. This was way easier and faster than a builder VM setup. Thank you.

anup-khanal, 2024-08-02:
I was following your tutorial above but then I got this error: **modprobe: ERROR: could not insert 'led_ugreen': Invalid argument** while running `modprobe -a i2c-dev led-ugreen ledtrig-oneshot ledtrig-netdev`
I have **TrueNAS Scale 24.04.2** running in **DXP6800 Pro**. I copied over the prebuilt **led-ugreen.ko** from [here](https://github.com/miskcoo/ugreen_dx4600_leds_controller/blob/truenas-build/build-scripts/truenas/build/TrueNAS-SCALE-Dragonfish/24.04.1.1)

sgreiner, 2024-08-02:
24.04.2 has a different kernel. so you need a suitable kernel module. you need to build it.

DavidePgr, 2024-08-07:
Hi, I'm too stupid for this conversation, can someone help me? I have the UG NAS 6800 Pro and I'm running TrueNAS Scale 24.04.2. All I want to do is to turn off ALL of the LEDs (power, LAN and disks). I tried following the guide but I must be missing something, so I deleted everything and now I want to start from scratch. Can someone tell me what the easiest way to achieve my goal is? Thank you so much🙏

sgreiner, 2024-08-07:
> Hi, I'm too stupid for this conversation, can someone help me? I have the UG NAS 6800 Pro and I'm running TrueNAS Scale 24.04.2. All I want to do is to turn off ALL of the LEDs (power, LAN and [...]

@DavidePgr let me try to give a summary:
1. compile the required kernel module for your **exact** truenas version. The easiest way is to compile it with the docker based script from this repo. Alternatively you can build it on a throwaway Builder VM like described above or ( this is not recommended!) on your production installation
2. install your production truenas
3. follow the guide above from the point "Target TrueNAS" onward step by step.
4. To turn the leds off you need to change the settings in scripts/ugreen-leds.conf (set all brightness to 0) before copying it to /etc/ugreen-leds.conf but you can also use this file to give the leds a nice color. The power led is not covered by this conf, but you could execute "echo 0 > /sys/class/leds/power/brightness " on startup. I recommend setting some nice colors and not turning them off.

If you have more specific questions I can try to help.

tooomm, 2024-08-07:
Can we accomplish the first step in your list as scheduled GitHub Actions run in the repo here?
Building the kmod for new TrueNas versions/kernels in Docker and providing the resulting file via a PR to the repo?

Would make live for many a bit easier 😀

DavidePgr, 2024-08-07:
> @DavidePgr let me try to give a summary:
>
> 1. compile the required kernel module for your **exact** truenas version. The easiest way is to compile it with the docker based script from this repo. [...]

Thanks, tomorrow I’ll have a look👍 is it correct that I have to do this for each truenas update?

Adrian-Grimm, 2024-08-09:
@DavidePgr I could recommend the [CLI solution](https://gist.github.com/Kerryliu/c380bb6b3b69be5671105fc23e19b7e8)

With that script: https://gist.github.com/Kerryliu/c380bb6b3b69be5671105fc23e19b7e8?permalink_comment_id=5109769#gistcomment-5109769 

With that you won't have trouble after upgrading TrueNAS - but I don't know how bad is to have to have i2c-dev enabled all the time...

DavidePgr, 2024-08-10:
I tried to run that script but I can't figure out a problem: I executed `chmod` on both `ugreen_leds_cli` and my `script.sh` file that are in the same directory. I loaded the i2c-dev module on boot but when I try to manually run the `script.sh` file it echoes "Ugreen LED CLI not found" even though it's in the same directory... I have no idea what I'm doing wrong

[image]

Can someone help? Thanks!

meyergru, 2024-08-10:
There can be several problems, IDK which apply to your situation:

1. If "." is not the the path (and it normally is not for security reasons), calling a script in the current directory must be carried out via ./script.sh.

2. Said script is written in bash syntax, not normal sh - this is evident from the firsr line of the script. Thus, '[[' is syntactically incorrect. You would have to call the script via "bash script.sh" instead of "script.sh".

3. You can also mount filesystems in a way that they do not allow programs to be executed from (mount option "noexec"), which may apply to your /mnt/home-ssd. In that case, no programs can be executed from there.

sgreiner, 2024-08-10:
@Adrian-Grimm @meyergru @DavidePgr This issue is about installing and running the solution of this repository on TrueNas with its kernel mod.
Please keep this issue clean and do not discuss off-topic like scripts from other repos or about learning unix basics.

KoTCiB, 2024-08-20:
Aloha,  Thank you for the nice solution.
1 - Would it be possible to provide similar control for the Power Led (Color and Intensity)?
2 - Please note that even when I set CHECK_LINK_SPEED=true, the color of the NETDEV Led is governed by COLOR_NETDEV_NORMAL.
Mahalo.

sgreiner, 2024-08-20:
@KoTCiB 

> 1 - Would it be possible to provide similar control for the Power Led (Color and Intensity)?

The color is not changed by the systemd services, so you can set it by: (copied from the main README of this project): 
```bash
echo 255 > /sys/class/leds/power/brightness    # non-zero brightness turns it on
echo "255 0 0" > /sys/class/leds/power/color   # set the color to RGB(255, 0, 0)
```

>2 - Please note that even when I set CHECK_LINK_SPEED=true, the color of the NETDEV Led is governed by COLOR_NETDEV_NORMAL.

Are you sure, that you used the correct name of the interface during systemd service creation from the systemd slice?
```bash
# change name of NIC if needed
systemctl start ugreen-netdevmon@enp88s0
systemctl enable ugreen-netdevmon@enp88s0
```
Maybe you put the cable into enp87s0 or enp89s0. In that case disable and stop the netdevmon on the wrong interface and use the right name after the @ symbol

KoTCiB, 2024-08-20:
Thank you very much.

P1tt187, 2024-10-12:
Thanks for your amazing work
With the documentation of this issue i managed to create an ansible playbook
I hope this will help the community

```yaml
- hosts: truenas
  tasks:
    - name: Detect current truenas version
      ansible.builtin.shell: cli -c 'system version_short'
      register: truenas_version

    - name: Remount boot-pool
      ansible.posix.mount:
        path: /
        fstype: zfs
        src: 'boot-pool/ROOT/{{ truenas_version.stdout }}'   
        opts: remount,rw
        state: mounted

    - name: Remount boot-pool usr
      ansible.posix.mount:
        path: /usr/
        fstype: zfs
        src: 'boot-pool/ROOT/{{ truenas_version.stdout }}/usr'   
        opts: remount,rw
        state: mounted 
        
    - name: Find apt-related binaries
      ansible.builtin.find:
        paths: /bin
        patterns: "apt*"
      register: apt_binaries
  
    - name: Set execute permissions on apt-related binaries
      ansible.builtin.file:
        path: "{{ item.path }}"
        mode: a+x
      loop: "{{ apt_binaries.files }}"
  
    - name: Set execute permissions on dpkg
      ansible.builtin.file:
          path: /usr/bin/dpkg
          mode: a+x  

    - name: Update apt cache
      environment:
        PATH: "/usr/bin:/usr/sbin"
      become: true
      ansible.builtin.apt:
        update_cache: yes
    
    - name: Install build-essential and linux-headers-generic
      environment:
        PATH: "/usr/bin:/usr/sbin"
      become: true
      ansible.builtin.apt:
        name:
          - build-essential
          - linux-headers-generic
        state: present
    - name: Ensure git repo is cloned
      ansible.builtin.git:
        repo: 'https://github.com/miskcoo/ugreen_dx4600_leds_controller.git'
        dest: /root/ugreen_dx4600_leds_controller
        force: true

    - name: Copy Build scripts
      ansible.builtin.copy:   
        remote_src: true
        src: "/root/ugreen_dx4600_leds_controller/build-scripts/debian/{{ item }}"
        dest: "/root/ugreen_dx4600_leds_controller/{{ item }}"
        mode: '0755'
      loop:
       - build-dkms-deb.sh
       - build-utils-deb.sh  

    - name: Build packages
      ansible.builtin.shell:
        chdir: /root/ugreen_dx4600_leds_controller/
        cmd: "bash {{ item }}"
      loop:
       - build-dkms-deb.sh
       - build-utils-deb.sh             
      environment:
        PATH: "/usr/bin:/usr/sbin"
        
    - name: Install packages
      ansible.builtin.apt:
        deb: "/root/ugreen_dx4600_leds_controller/{{ item }}"
        state: present
      loop:
        - led-ugreen-dkms.deb
        - led-ugreen-utils.deb  
      environment:
        PATH: "/usr/bin:/usr/sbin"  
    
    - name: Enable services
      ansible.builtin.systemd_service:
        daemon_reload: true
        enabled: true
        name: "{{ item }}"
        state: "started"
      loop:
        - ugreen-diskiomon
        - ugreen-netdevmon@br0  

```

sgreiner, 2024-10-29:
updated https://github.com/miskcoo/ugreen_leds_controller/issues/13#issue-2363286910 for TrueNAS-SCALE-ElectricEel 24.10.0
(just paths and version numbers)

rybackisback, 2024-12-01:
Ive managed to get it to run. I notice that my power led just keeps blinking. Where do I make changes as per the readme file for power led? is that in the ugreen-led.conf? Also my DXP 8800 Plus has 2 nics, how do i activate both of them? i repeated the netdevmon command but the led only seems to be picking up activity for one nic.

0x556c79, 2024-12-01:
> Ive managed to get it to run. I notice that my power led just keeps blinking. Where do I make changes as per the readme file for power led? is that in the ugreen-led.conf? Also my DXP 8800 Plus has [...]

Yes, I have the same problem. It would be really nice to be able to adjust the behavior of the power LED.

Also, "ugreen-check-standby" doesn't work properly. The color of the LEDs always changes immediately even though the hard disks are active and definitely not in standby.

miskcoo, 2024-12-01:
Hi, this issue is a guide for installing modules in truenas. Please create
a new issue for other problems / bugs.

For the issue about "ugreen-check-standby", can you tell me your HDD model
(in a new issue)?

rybackisback, 2024-12-01:
Thanks @miskcoo will open a new issue.

0x556c79, 2024-12-01:
Hey @miskcoo I have created a small bash script that simplifies the installation for TrueNAS. You may want to link to it in the instructions above or add it to the repository.

```
#!/bin/bash

# Ensure script is run as root
if [ "$EUID" -ne 0 ]; then
    echo "Please run as root."
    exit 1
fi

# Variables
REPO_URL="https://github.com/miskcoo/ugreen_leds_controller/raw/refs/heads/gh-actions/build-scripts/truenas/build"
SUPPORTED_VERSIONS=("24.10.0" "24.10.0.1" "24.10.0.2" "24.04.0" "24.04.1" "24.04.1.1" "24.04.2")
TRUENAS_VERSION=$(cat /etc/version | grep -oP '^[0-9]+\.[0-9]+(\.[0-9]+)?(\.[0-9]+)?')
TRUENAS_SERIES=$(echo "$TRUENAS_VERSION" | cut -d'.' -f1,2)

# Map version to TrueNAS name
if [[ "$TRUENAS_SERIES" == "24.10" ]]; then
    TRUENAS_NAME="TrueNAS-SCALE-ElectricEel"
elif [[ "$TRUENAS_SERIES" == "24.04" ]]; then
    TRUENAS_NAME="TrueNAS-SCALE-Dragonfish"
else
    echo "Unsupported TrueNAS SCALE version series: ${TRUENAS_SERIES}."
    echo "Please build the kernel module manually."
    exit 1
fi

# Validate full version
if [[ ! " ${SUPPORTED_VERSIONS[@]} " =~ " ${TRUENAS_VERSION} " ]]; then
    echo "Unsupported TrueNAS SCALE version: ${TRUENAS_VERSION}."
    echo "Please build the kernel module manually."
    exit 1
fi

# Construct the kernel module URL
MODULE_URL="${REPO_URL}/${TRUENAS_NAME}/${TRUENAS_VERSION}/led-ugreen.ko"

# Test if the kernel module URL is valid
echo "Checking if kernel module exists for TrueNAS version ${TRUENAS_VERSION}..."
if ! curl --head --silent --fail "${MODULE_URL}" > /dev/null; then
    echo "Kernel module not found for TrueNAS version ${TRUENAS_VERSION}."
    echo "Please build the kernel module manually."
    exit 1
fi

# Variables for remount paths
BOOT_POOL_PATH="boot-pool/ROOT/${TRUENAS_VERSION}"

# Step 1: Remount boot-pool datasets with write access
echo "Remounting boot-pool datasets with write access..."
mount -o remount,rw "${BOOT_POOL_PATH}/usr" || exit 1
mount -o remount,rw "${BOOT_POOL_PATH}/etc" || exit 1

# Step 2: Clone the Ugreen LEDs Controller repository
echo "Cloning Ugreen LEDs Controller repository..."
cd /root
if [ ! -d "/root/ugreen_leds_controller" ]; then
    git clone https://github.com/miskcoo/ugreen_leds_controller.git || exit 1
fi

# Step 3: Download and install the kernel module
echo "Downloading and installing the kernel module..."
mkdir -p "/lib/modules/$(uname -r)/extra"
curl -o "/lib/modules/$(uname -r)/extra/led-ugreen.ko" "${MODULE_URL}" || exit 1
chmod 644 "/lib/modules/$(uname -r)/extra/led-ugreen.ko"

# Step 4: Create kernel module load configuration
echo "Creating kernel module load configuration..."
cat <<EOL > /etc/modules-load.d/ugreen-led.conf
i2c-dev
led-ugreen
ledtrig-oneshot
ledtrig-netdev
EOL
chmod 644 /etc/modules-load.d/ugreen-led.conf

# Step 5: Load kernel modules
echo "Loading kernel modules..."
depmod
modprobe -a i2c-dev led-ugreen ledtrig-oneshot ledtrig-netdev

# Step 6: Ask user if they want to modify the configuration file
echo "Do you want to modify the LED configuration file now? (y/n)"
read -r MODIFY_CONF
if [[ "$MODIFY_CONF" == "y" ]]; then
    nano /root/ugreen_leds_controller/scripts/ugreen-leds.conf
fi

# Copy the configuration file
cp /root/ugreen_leds_controller/scripts/ugreen-leds.conf /etc/ugreen-leds.conf
chmod 644 /etc/ugreen-leds.conf

# Step 7: Determine the network interface name automatically
NETWORK_INTERFACE=$(ifconfig | grep -oP 'enp[0-9]+[a-z][0-9]+')
if [ -z "$NETWORK_INTERFACE" ]; then
    echo "Failed to detect the network interface name."
    exit 1
fi
echo "Detected network interface: ${NETWORK_INTERFACE}"

# Step 8: Copy scripts and configure services
echo "Setting up systemd services..."
cd /root/ugreen_leds_controller

scripts=("ugreen-diskiomon" "ugreen-netdevmon" "ugreen-probe-leds")
for script in "${scripts[@]}"; do
    chmod +x "scripts/$script"
    cp "scripts/$script" /usr/bin
done

cp scripts/*.service /etc/systemd/system/
systemctl daemon-reload

# Enable and start services
echo "Enabling and starting services..."
systemctl start ugreen-diskiomon.service
systemctl enable ugreen-diskiomon.service
systemctl start "ugreen-netdevmon@${NETWORK_INTERFACE}"
systemctl enable "ugreen-netdevmon@${NETWORK_INTERFACE}"

echo "Setup complete. Reboot your system to verify."
```

deepybee, 2024-12-07:
Just used your script @0x556c79, worked a treat, thanks

rybackisback, 2024-12-08:
> Just used your script @0x556c79, worked a treat, thanks

Can someone please provide some advice how best to run the script above in truenas? I tried to setup a cron job but it fails.

0x556c79, 2024-12-08:
> > Just used your script @0x556c79, worked a treat, thanks
> 
> Can someone please provide some advice how best to run the script above in truenas? I tried to setup a cron job but it fails.

Why a cron job?

The script is intended for installation and only needs to be run once and again after each update.
Simply log in via SSH and create a file in the user's home directory, e.g. nano install-ugreen-leds-controller.sh.
Copy the script into the file and save it.
Execute it with sudo.
Run it once after every system update.
If you customize the configuration, I would also save the customizations somewhere so that you can simply copy in your own configuration during the installation.

rybackisback, 2024-12-08:
> Why a cron job?
>
> The script is intended for installation and only needs to be run once and again after each update. Simply log in via SSH and create a file in the user's home directory, e.g. nano [...]

Thanks for the explanation! I am new to truenas.

0x556c79, 2024-12-14:
> > > > Just used your script @0x556c79, worked a treat, thanks
> > > 
> > > 
> > > Can someone please provide some advice how best to run the script above in truenas? I tried to setup a cron job but it fails.
> > 
> > 
> > Why a cron job?
> > The script is intended for installation and only needs to be run once and again after each update. Simply log in via SSH and create a file in the user's home directory, e.g. nano install-ugreen-leds-controller.sh. Copy the script into the file and save it. Execute it with sudo. Run it once after every system update. If you customize the configuration, I would also save the customizations somewhere so that you can simply copy in your own configuration during the installation.
> 
> Thanks for the explanation! I am new to truenas. My nic activity led stay solid and does not blink to reflect with network activity. Not sure what would cause that?

Is the ugreen-netdevmon service for this running?

`systemctl status “ugreen-netdevmon@enp0s” `

Replace the name of the network interface with what it is called in your system
You can display it with `ifconfig`

0x556c79, 2024-12-14:
> Thanks for the explanation! I am new to truenas. My nic activity led stay solid and does not blink to reflect with network activity. Not sure what would cause that?

Hey @rybackisback, I had an error in the script. It was not taken into account that there can be more than one interface and so the systemd service was created incorrectly and does not start. I have adjusted my comment above with the script. Test it again now.

rybackisback, 2024-12-17:
> > > > > Just used your script @0x556c79, worked a treat, thanks
> > > > 
> > > > 
> > > > Can someone please provide some advice how best to run the script above in truenas? I tried to setup a cron job but it fails.
> > > 
> > > 
> > > Why a cron job?
> > > The script is intended for installation and only needs to be run once and again after each update. Simply log in via SSH and create a file in the user's home directory, e.g. nano install-ugreen-leds-controller.sh. Copy the script into the file and save it. Execute it with sudo. Run it once after every system update. If you customize the configuration, I would also save the customizations somewhere so that you can simply copy in your own configuration during the installation.
> > 
> > 
> > Thanks for the explanation! I am new to truenas. My nic activity led stay solid and does not blink to reflect with network activity. Not sure what would cause that?
> 
> Hey @rybackisback, I had an error in the script. It was not taken into account that there can be more than one interface and so the systemd service was created incorrectly and does not start. I have updated the script and also created a repo for it, feel free to test it again.

Brilliant!..thanks! that worked.

miskcoo, 2025-01-13:
> https://github.com/0x556c79/install-ugreen-leds-controller

@0x556c79 linked it in readme.

rybackisback, 2025-01-18:
@0x556c79 im struggling to follow what changes do i need to make to the curl install command to ensure compatibility with truenas version 24.10.1? currently when i follow the instructions it comes up withe the message "please build kernel module manually"

0x556c79, 2025-01-18:
> Hi [@0x556c79](https://github.com/0x556c79) im struggling to follow what changes do i need to make to the curl install command to ensure compatibility with truenas version 24.10.1? currently when i follow the instructions it comes up withe the message "please build kernel module manually"

You have to use -v to specify the version of TrueNAS for which a kernel module is available. In this case 24.10.0.2
For 24.10.1 there is none in the repository but the one from the previous version works without problems as both use the same kernel version.

./install_ugreen_leds_controller.sh -v 24.10.0.2

rybackisback, 2025-01-18:
thanks @0x556c79 i ran ./install_ugreen_leds_controller.sh -v 24.10.0.2 but now get the following:  cp: cannot stat 'scripts/*.service': No such file or directory

miskcoo, 2025-01-18:
> thanks [@0x556c79](https://github.com/0x556c79) i ran ./install_ugreen_leds_controller.sh -v 24.10.0.2 but now get the following: cp: cannot stat 'scripts/*.service': No such file or directory

The latest PR (#51) moved the systemd services to `scripts/systemd/*.service`. Maybe a small change for the script is needed.

0x556c79, 2025-01-18:
> > thanks [@0x556c79](https://github.com/0x556c79) i ran ./install_ugreen_leds_controller.sh -v 24.10.0.2 but now get the following: cp: cannot stat 'scripts/*.service': No such file or directory
> 
> The latest PR ([#51](https://github.com/miskcoo/ugreen_leds_controller/pull/51)) moved the systemd services to `scripts/systemd/*.service`. Maybe a small change for the script is needed.
> 
> You can try to change [line 257](https://github.com/0x556c79/install_ugreen_leds_controller/blob/main/install_ugreen_leds_controller.sh#L257) to `cp scripts/systemd/*.service /etc/systemd/system/`.

Script is updated @rybackisback

rybackisback, 2025-01-19:
thanks! @0x556c79 and @miskcoo that has fixed the error..everything working as expected

Ladduro, 2025-01-19:
Anyone willing to compile the kernel module for 24.10.1?

miskcoo, 2025-01-19:
> Anyone willing to compile the kernel module for 24.10.1? I am getting this when I want to build it via docker
> 
> + wget https://download.truenas.com/TrueNAS-SCALE-ElectricEel/24.10.1/packages/Packages.gz
> --2025-01-19 09:03:41--  https://download.truenas.com/TrueNAS-SCALE-ElectricEel/24.10.1/packages/Packages.gz
> Resolving download.truenas.com (download.truenas.com)... 185.244.226.2
> Connecting to download.truenas.com (download.truenas.com)|185.244.226.2|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2025-01-19 09:03:42 ERROR 404: Not Found.

Please try the previous version `24.10.0.2` when encountering such errors. See #28.

Ladduro, 2025-01-19:
The module for 24.10.0.2 does not work unfortunately 

```shell
# depmod
depmod: ERROR: failed to load symbols from /lib/modules/6.6.44-production+truenas/extra/led-ugreen.ko: Exec format error
```

Ladduro, 2025-01-30:
Hi all, 

I am still getting this error while using a prebuild kernel module

```shell
truenas_admin@saturn[~]$ sudo modprobe -a led-ugreen
modprobe: ERROR: could not insert 'led_ugreen': Exec format error
``` 
I am on 24.10.2 and downloaded the module from Alternative 0 point. 

Any ideas?

sgreiner, 2025-01-30:
> Hi all,
> 
> I am still getting this error while using a prebuild kernel module
> 
> truenas_admin@saturn[~]$ sudo modprobe -a led-ugreen
> modprobe: ERROR: could not insert 'led_ugreen': Exec format error
> 
> I am on 24.10.2 and 6800 Pro, downloaded the module from Alternative 0 point.
> 
> Any ideas?

@Ladduro just upgraded my system to 24.10.2 - the prebuilt kmod works fine.
Please check if you downloaded the correct file and the files integrity.

sha256sum /lib/modules/6.6.44-production+truenas/extra/led-ugreen.ko
9e708b5d0bb47881cd396e63d89e6f399f94151e470ca572fa4fdf428fa84655

I also updated the instructions above a little bit regarding the new systemd unit (ugreen-probe-leds.service) and folder structure.

closed 2025-01-30 by sgreiner

Ryands991, 2025-02-01:
> [@Ladduro](https://github.com/Ladduro) just upgraded my system to 24.10.2 - the prebuilt kmod works fine. Please check if you downloaded the correct file and the files integrity.

I am having the same issue, file integrity passed check.

miskcoo, 2025-02-03:
@Ryands991 @Ladduro 

You can also check the output of the following command to confirm that you put `led-ugreen.ko` in the correct directory.
```
modinfo /lib/modules/$(uname -r)/extra/led-ugreen.ko
```
In the last line you should see something like below:
```
license:        GPL v2
description:    UGREEN NAS LED driver
alias:          i2c:led-ugreen
depends:
retpoline:      Y
name:           led_ugreen
vermagic:       6.6.44-production+truenas (<-- this is your truenas kernel version, for 24.10.2) SMP preempt mod_unload modversions
parm:           verbose:Enable verbose output (bool)
```

Ryands991, 2025-02-15:
> [@Ryands991](https://github.com/Ryands991) [@Ladduro](https://github.com/Ladduro)
>
> You can also check the output of the following command to confirm that you put `led-ugreen.ko` in the correct [...]

Output is still relatively the same.

truenas_admin@truenas[~]$ modinfo /lib/modules/$(uname -r)/extra/led-ugreen.ko      
filename:       /lib/modules/6.6.44-production+truenas/extra/led-ugreen.ko
modinfo: ERROR: could not get modinfo from 'led_ugreen': Exec format error

pjrobot, 2026-10-19:
For anyone on TrueNAS SCALE tired of rebuilding the module after every update: I packaged a userspace alternative as a TrueNAS Custom App.

https://github.com/pjrobot/ugreen-leds-truenas

- No kernel module. It speaks the controller's I2C protocol (as documented in this repo, thanks @miskcoo) through stock `i2c-dev`, so nothing depends on the kernel version and nothing lives on the read-only root.
- One YAML paste into Apps → Custom App; the daemon is embedded in the compose file. The only other step is a Pre Init task running `modprobe i2c-dev`.
- Behaves like UGOS: power steady, network blinks on traffic, each bay steady with a disk present and blinking on I/O, red while a pool is not ONLINE. Blinking uses the controller's hardware blink mode, so it isn't chatty on the bus.
- Tested on a DXP4800 Plus with TrueNAS 25.04.2.6. Other models should work (bay order is configurable); reports welcome.

The script is standalone Python with no dependencies, so it also runs as a plain systemd unit on Debian/Proxmox.