miskcoo/ugreen_leds_controller #19: DXP2800

Balkoth, 2024-06-25:
Does this work on the DXP2800?

miskcoo, 2024-06-25:
I don't know since I don't have such a device. You can test it if you have one.

Logian, 2024-06-27:
Can confirm that the command-line tool is working and LEDs are fully configurable. Naturally, there's only leds power, netdev, disk1 and disk2 available.

miskcoo, 2024-06-27:
@Logian can you check whether the product name `dmidecode --string system-product-name` starts with `DXP2800`? I can add this model to the scripts so that you won't see some warning that your device is untested.

Logian, 2024-06-29:
@miskcoo Output of dmidecode is in fact "DXP2800". Nothing more, nothing less.

aqsede, 2024-07-03:
I'm lacking in knowledge to enable this. 

I am now able to build the cli on my OMV with debian12. However, still stuck at enabling the LED so it work like normal.
the dkms has errors with kernel headers. So i'm stuck there.
```
nas@omvnas:/home/ugnas/ugreen_dx4600_leds_controller$ sudo dkms add -m led-ugreen -v 0.1
Creating symlink /var/lib/dkms/led-ugreen/0.1/source -> /usr/src/led-ugreen-0.1
nas@omvnas:/home/ugnas/ugreen_dx4600_leds_controller$ sudo dkms build -m led-ugreen -v 0.1
Sign command: /lib/modules/6.1.0-22-amd64/build/scripts/sign-file
Binary /lib/modules/6.1.0-22-amd64/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.1.0-22-amd64 cannot be found at /lib/modules/6.1.0-22-amd64/build or /lib/modules/6.1.0-22-amd64/source.
Please install the linux-headers-6.1.0-22-amd64 package or use the --kernelsourcedir option to tell DKMS where it's located.
```

miskcoo, 2024-07-03:
> Please install the linux-headers-6.1.0-22-amd64 package or use the --kernelsourcedir option to tell DKMS where it's located.

@aqsede Hi, do you forget to install this `apt install linux-headers-6.1.0-22-amd64`?

closed 2024-09-18 by miskcoo

0n1cOn3, 2025-05-07:
I can confirm that it works flawlessly with the DXP2800 :D 

https://github.com/user-attachments/assets/8e8a91eb-4362-401e-b845-20b40d06c71b