miskcoo/ugreen_leds_controller #21: UGREEN DXP2800 lights still on after drives spin down

kraizelburg, 2024-07-12:
Hi, first of all thank you for providing the driver. 

I just wanted to mention that this morning, after my disks had spin down the lights were still on, no off nor pulsing like in regular ugreen, als logs show the following warnings.

[image]

Ideally the lights would off if drive is spin down or pulsin like with factory ugreen.

Thanks in advance

kraizelburg, 2024-07-12:
To be honest I am not an expert in driver settings and I have no clue where to change this. Do you know where is this config file located in unraid plugin?

dieneuser, 2024-07-12:
> To be honest I am not an expert in driver settings and I have no clue where to change this. Do you know where is this config file located in unraid plugin?

Your question has long been answered in the Unraid forum, where you first described the issue: https://forums.unraid.net/topic/168423-ugreen-nas-led-control/?do=findComment&comment=1436097

ich777, 2024-07-15:
> To be honest I am not an expert in driver settings and I have no clue where to change this. Do you know where is this config file located in unraid plugin?

I already pushed an update which should fix the issue that you see in your syslog, are you already on the latest plugin version?

As said for the second thing that the LEDs stay on, this is expected with the current implementation.

kraizelburg, 2024-07-15:
> I already pushed an update which should fix the issue that you see in your syslog, are you already on the latest plugin version?
>
> As said for the second thing that the LEDs stay on, this is expected [...]

Hi, I am on version 2024.07.12 which I believe is the latest one. So far I can confirm that I have not seen the error I posted in the logs. Regarding the LEDs on even after disks spun down, I understand it is the expected behavior for now. Thank you in advance.

ich777, 2024-07-15:
@miskcoo I think this can be closed for now (just to keep your Issues clean :) ).

closed 2024-07-16 by miskcoo

kraizelburg, 2024-07-16:
Hi, this morning I had a look at the logs and I saw this: not it is LED 2 nad LED 3
[image]

ich777, 2024-07-16:
@kraizelburg please post your Diagnostics.
Are the LEDs still working?

kraizelburg, 2024-07-16:
Yes LEDs are working as before no issues there, I just don’t know about these warnings. I will post diagnostics when I get home.

kraizelburg, 2024-07-16:
Please find diagnostics. Thanks

[dxp2800-diagnostics-20240716-1230.zip](https://github.com/user-attachments/files/16247365/dxp2800-diagnostics-20240716-1230.zip)

miskcoo, 2024-07-16:
maybe it is the same problem. A workaround is to increase the blink interval of disk leds, but i think dxp2800 is slightly different from other models, so the correct solution may be adjusting the sleep interval in the driver. Please wait for some days, i will check that this or the next weekend.

nevertheless, this warning is not critical.

kraizelburg, 2024-07-16:
> maybe it is the same problem. A workaround is to increase the blink interval of disk leds, but i think dxp2800 is slightly different from other models, so the correct solution may be adjusting the [...]

Yes it is a warning only and I see LEDs are working so I don't really understand what is the point of this warning. 
Thank you