miskcoo/ugreen_leds_controller #24: ugreen_leds_cli: Segmentation fault when turning on blink or breath for exactly 3 leds

derandiunddasbo, 2024-09-27:
On my UGreen DXP2800 the ugreen_leds_cli command line tool crashes with a Segmentation fault, when turning on blink or breath for exactly three or five leds in one call. All other combinations seem to work fine:

```
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 disk3 disk4 disk5 disk6 disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 disk3 disk4 disk5 disk6 disk7 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 disk3 disk4 disk5 disk6 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 disk3 disk4 disk5 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 disk3 disk4 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 disk3 -blink 2000 50
Segmentation fault
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 disk2 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev disk1 -blink 2000 50
Segmentation fault
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power netdev -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli power -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli netdev disk1 disk2 disk3 disk4 disk5 disk6 disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk1 disk2 disk3 disk4 disk5 disk6 disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk2 disk3 disk4 disk5 disk6 disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk3 disk4 disk5 disk6 disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk4 disk5 disk6 disk7 disk8 -blink 2000 50
Segmentation fault
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk5 disk6 disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk6 disk7 disk8 -blink 2000 50
Segmentation fault
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk7 disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk8 -blink 2000 50
root@proxmoxn2:/usr/local/bin# ugreen_leds_cli disk8 -blink 2000 50
```

miskcoo, 2024-09-27:
fixed, you can download the release again.

derandiunddasbo, 2024-09-27:
Yes, it's working now. Thank you for the quick fix!

closed 2024-09-27 by derandiunddasbo