miskcoo/ugreen_leds_controller #25: Arch - DXP4800 Plus

xerootg, 2024-10-02:
I have this working on arch. I've got some fun stuff in dmesg though. I'll take the readme I wrote up and script it as a contribution, possibly even as a PKGBUILD so it can be handled as an AUR if I have time, if this isn't concerning (it seems concerning). [arch install readme](https://github.com/xerootg/ugreen_dx4600_leds_controller/blob/arch/build-scripts/arch.md)
```log
[    7.488157] i2c_dev: i2c /dev entries driver
[    7.499456] led_ugreen: loading out-of-tree module taints kernel.
[    7.499462] led_ugreen: module verification failed: signature and/or required key missing - tainting kernel
[    7.499699] led_ugreen: initializing
[    7.504933] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:1f.4/i2c-9/9-003a'
[    7.504939] CPU: 4 UID: 0 PID: 542 Comm: ugreen-probe-le Tainted: G           OE      6.11.1-zen1-1-zen #1 263364632483c3101e3c22db458ac664e17f4493
[    7.504943] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[    7.504944] Hardware name: UGREEN DXP4800 Plus/Default string, BIOS 5.27 02/26/2024
[    7.504945] Call Trace:
[    7.504948]  <TASK>
[    7.504950]  dump_stack_lvl+0x5d/0x80
[    7.504956]  sysfs_warn_dup.cold+0x17/0x23
[    7.504958]  sysfs_create_dir_ns+0xce/0xe0
[    7.504962]  kobject_add_internal+0x1c2/0x3f0
[    7.504966]  kobject_add+0x105/0x140
[    7.504968]  ? __kmalloc_cache_noprof+0x11c/0x310
[    7.504970]  device_add+0xe2/0xbd0
[    7.504973]  ? complete_all+0x24/0xa0
[    7.504976]  i2c_new_client_device+0x27e/0x3d0
[    7.504979]  new_device_store+0x112/0x220
[    7.504981]  kernfs_fop_write_iter+0x13e/0x1f0
[    7.504983]  vfs_write+0x31d/0x4a0
[    7.504986]  __x64_sys_write+0x72/0xf0
[    7.504988]  do_syscall_64+0x82/0x190
[    7.504991]  ? exc_page_fault+0x81/0x190
[    7.504993]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[    7.504996] RIP: 0033:0x78d4e28957a4
[    7.505013] Code: c7 00 16 00 00 00 b8 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 80 3d c5 28 0e 00 00 74 13 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 54 c3 0f 1f 00 55 48 89 e5 48 83 ec 20 48 89
[    7.505014] RSP: 002b:00007ffd61d91c58 EFLAGS: 00000202 ORIG_RAX: 0000000000000001
[    7.505017] RAX: ffffffffffffffda RBX: 0000000000000010 RCX: 000078d4e28957a4
[    7.505019] RDX: 0000000000000010 RSI: 00005bc6d0117280 RDI: 0000000000000001
[    7.505020] RBP: 00007ffd61d91c80 R08: 0000000000000073 R09: 0000000000000000
[    7.505021] R10: 0000000000000000 R11: 0000000000000202 R12: 0000000000000010
[    7.505022] R13: 00005bc6d0117280 R14: 000078d4e29715c0 R15: 000078d4e296eea0
[    7.505024]  </TASK>
[    7.505025] kobject: kobject_add_internal failed for 9-003a with -EEXIST, don't try to register things with the same name in the same directory.
[    7.505083] led_ugreen: i2c probed
[    7.506070] i2c i2c-9: Failed to register i2c client led-ugreen at 0x3a (-17)
[    7.547115] led_ugreen: probed led id 0, status 1, rgb 0xffffff, brightness 128, t_on 256, t_cycle 768
[    7.551933] led_ugreen: probed led id 1, status 1, rgb 0xffa500, brightness 255, t_on 256, t_cycle 4864
[    7.568887] led_ugreen: probed led id 2, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    7.579439] led_ugreen: probed led id 3, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    7.590497] led_ugreen: probed led id 4, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    7.602291] led_ugreen: probed led id 5, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    8.195121] i2c i2c-9: new_device: Instantiated device led-ugreen at 0x3a
```

miskcoo, 2024-10-02:
It seems the message comes from the duplicated call of `ugreen-probe-leds` in both `ugreen-diskiomon` and `ugreen-netdevmon`. 
Can you try to only enable one of them, and check whether this message will disappear? If so, then try to start another service and check whether the message appear again.

xerootg, 2024-10-02:
oh, that last commit seems to have atleast killed the error:
```bash
[    7.395324] i2c_dev: i2c /dev entries driver
[    7.405321] led_ugreen: loading out-of-tree module taints kernel.
[    7.405328] led_ugreen: module verification failed: signature and/or required key missing - tainting kernel
[    7.405561] led_ugreen: initializing
[    7.412167] led_ugreen: i2c probed
[    7.416055] led_ugreen: probed led id 0, status 1, rgb 0xffffff, brightness 128, t_on 256, t_cycle 768
[    7.420299] led_ugreen: probed led id 1, status 1, rgb 0xffa500, brightness 255, t_on 256, t_cycle 4864
[    7.473987] led_ugreen: probed led id 2, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    7.484873] led_ugreen: probed led id 3, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    7.531837] led_ugreen: probed led id 4, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    7.541828] led_ugreen: probed led id 5, status 1, rgb 0xffffff, brightness 255, t_on 256, t_cycle 4864
[    8.121078] i2c i2c-10: new_device: Instantiated device led-ugreen at 0x3a
```
I just need to figure out signing the dkms produced module and I'd consider this absolutely a working solution! :)

xerootg, 2024-10-02:
actually, it looks like I'm missing something for ugreen_leds_cli
```bash
[xero@nas ~]$ sudo ugreen_leds_cli all
Err: fail to open the I2C device.
Please check that (1) you have the root permission;
              and (2) the i2c-dev module is loaded.
```
strace:
```
getdents64(3, 0x34dded00 /* 13 entries */, 32768) = 400
newfstatat(AT_FDCWD, "/sys/class/i2c-dev/i2c-3", {st_mode=S_IFDIR|0755, st_size=0, ...}, 0) = 0
openat(AT_FDCWD, "/sys/class/i2c-dev/i2c-3/device/name", O_RDONLY) = 4
read(4, "i915 gmbus tc1\n", 8191)       = 15
close(4)                                = 0
newfstatat(AT_FDCWD, "/sys/class/i2c-dev/i2c-10", {st_mode=S_IFDIR|0755, st_size=0, ...}, 0) = 0
openat(AT_FDCWD, "/sys/class/i2c-dev/i2c-10/device/name", O_RDONLY) = 4
read(4, "SMBus I801 adapter at efa0\n", 8191) = 27
openat(AT_FDCWD, "/dev/i2c-10", O_RDWR) = 5
ioctl(5, _IOC(_IOC_NONE, 0x7, 0x3, 0), 0x3a) = -1 EBUSY (Device or resource busy)
close(5)                                = 0
close(4)                                = 0
close(3)                                = 0
```

I'm running arch linux-zen 6.11, and i2c_dev is loaded:
```bash
[xero@nas ~]$ lsmod | grep i2c
i2c_dev                28672  0
i2c_i801               40960  0
i2c_smbus              20480  1 i2c_i801
i2c_mux                16384  1 i2c_i801
i2c_algo_bit           24576  2 xe,i915
```

miskcoo, 2024-10-02:
@xerootg Once the driver is loaded, `ugreen_leds_cli` can no longer controls the lights -- you should directly write files in `/sys/class/leds/`; or unload the driver.

xerootg, 2024-10-02:
I wondered if that was a thing. I assume it would be a reasonable thing to check if the driver is loaded and either use the driver or exit with a more concise error message?

miskcoo, 2024-10-03:
Of course it is reasonable. There is some unmerged code for this, and I can merge them recently.