miskcoo/ugreen_leds_controller #28: Missing kernel builds for newer TrueNAS versions

tooomm, 2024-10-13:
Looking at https://download.truenas.com/, where the `build-all.sh` script checks as well, there are versions listed up to `24.04.2.3`:

[image]

The [latest action runs](https://github.com/miskcoo/ugreen_dx4600_leds_controller/actions/workflows/build-truenas.yml) do only build an old release from one month ago:
`Building TrueNAS-SCALE-Dragonfish/24.04.2.1`

The file storage even only has `24.04.2` 🤔 
https://github.com/miskcoo/ugreen_dx4600_leds_controller/tree/gh-actions/build-scripts/truenas/build/TrueNAS-SCALE-Dragonfish

miskcoo, 2024-10-15:
Unfortunately, I am not a TrueNAS user and I cannot find the _linux-headers_ package in the download link. It seems there is something different compared to other version.

Maybe the old module (24.04.2) also works for these versions?

tooomm, 2024-10-19:
I ran the [command from the script](https://github.com/miskcoo/ugreen_leds_controller/blob/master/build-scripts/truenas/build-all.sh) locally, and it does find all versions correctly:

```
$ curl https://download.truenas.com/ | grep -Po "(?<=href=\")TrueNAS-SCALE-\w*?/\d+(\.\d+)*" | uniq
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 6084k  100 6084k    0     0  2857k      0  0:00:02  0:00:02 --:--:-- 2860k
TrueNAS-SCALE-Angelfish/22.02.0
TrueNAS-SCALE-Angelfish/22.02.0.1
TrueNAS-SCALE-Angelfish/22.02.1
TrueNAS-SCALE-Angelfish/22.02.2
TrueNAS-SCALE-Angelfish/22.02.2.1
TrueNAS-SCALE-Angelfish/22.02.3
TrueNAS-SCALE-Angelfish/22.02.4
TrueNAS-SCALE-Bluefin/22.12.0
TrueNAS-SCALE-Bluefin/22.12.1
TrueNAS-SCALE-Bluefin/22.12.2
TrueNAS-SCALE-Bluefin/22.12.3
TrueNAS-SCALE-Bluefin/22.12.3.1
TrueNAS-SCALE-Bluefin/22.12.3.2
TrueNAS-SCALE-Bluefin/22.12.3.3
TrueNAS-SCALE-Bluefin/22.12.4
TrueNAS-SCALE-Bluefin/22.12.4.1
TrueNAS-SCALE-Bluefin/22.12.4.2
TrueNAS-SCALE-Cobia/23.10.0
TrueNAS-SCALE-Cobia/23.10.0.1
TrueNAS-SCALE-Cobia/23.10.1
TrueNAS-SCALE-Cobia/23.10.1.1
TrueNAS-SCALE-Cobia/23.10.1.3
TrueNAS-SCALE-Cobia/23.10.2
TrueNAS-SCALE-Dragonfish/24.04.0
TrueNAS-SCALE-Dragonfish/24.04.1
TrueNAS-SCALE-Dragonfish/24.04.1.1
TrueNAS-SCALE-Dragonfish/24.04.2
TrueNAS-SCALE-Dragonfish/24.04.2.1
TrueNAS-SCALE-Dragonfish/24.04.2.2
TrueNAS-SCALE-Dragonfish/24.04.2.3
```

<br>

Maybe the logs of the action give a better hint what might goes wrong here.

miskcoo, 2024-10-19:
Sure, it will find these versions. 

For those older than 24.04, I manually excluded them in the next few lines, so it is expected they are not built.

But for 24.04.2.x, the script failed at finding the `linux-headers` package. For example, you can find it in the packages subdirectory of [24.04.2](https://download.truenas.com/TrueNAS-SCALE-Dragonfish/24.04.2/packages/linux-headers-truenas-production-amd64_6.6.32-production+truenas-1_amd64.deb)), but it doesn't exist in 24.04.2.x.

sgreiner, 2024-10-22:
24.04.2 and 24.04.2.1, 24.04.2.2, 24.04.2.3 use all the same Linux kernel version 6.6.32, a build would be redundant, the kmod build for 24.04.2 can be used for 24.04.2.3.

tooomm, 2024-10-24:
Thanks for noting sgreiner, maybe that also explains the missing header package.

With the new 24.10 release later this month, all should be fine again :)

deepybee, 2024-10-29:
FWIW, and apologies to @tooomm for the slight hijack, compiling from source works flawlessly on Electric Eel RC2, thanks so much for this repo @miskcoo, I had no idea how much the scrolling LEDs on my 8800 would bother me until I saw them working beautifully as they should do!

Excellent work and truly thankful here. 👏 🙇

deepybee, 2024-10-29:
Important update: just updated from RC2 to the 24.10.0 release (which is available as an update if you're running the RC, but does not yet seem to be publicised according to the website and TN Discord) and it seems the update process overwrites all of 

```
/etc/
/etc/modprobe.d/
/etc/modules-load.d/
/lib/modules/$(uname -r)/kernel/drivers/leds/
```

and mounts a read only filesystem again, so [developer mode](https://www.truenas.com/docs/scale/scaletutorials/systemsettings/advanced/developermode/) must be enabled to install the kernel module and associated config files to have it load on boot.

I have raise #35 to update the repo README.

sgreiner, 2024-10-29:
@deepybee this is totally expected for TrueNAS. As you can read in the Docs the Readme already links to.
see https://github.com/miskcoo/ugreen_leds_controller/issues/13 
_"After a TrueNAS Scale Update you need to repeat all steps, as the boot-pool datasets are versioned and our changes will be gone in a new boot-pool dataset."_

deepybee, 2024-10-29:
@sgreiner yeah I just caught that now, thanks. Have closed my PR.

miskcoo, 2024-10-30:
`24.10` has been successfully built. I thought we can close this issue now.

closed 2024-10-30 by miskcoo