miskcoo/ugreen_leds_controller #31: Build and install failure on Debian 12.7 / Linux `6.1.0-26-amd64`

tigattack, 2024-10-16:
Hi, thanks for this project! Awesome work and super useful.

I had it installed before, and it worked great, but after updating it seems to have failed.

First of all, I noticed this in the logs of the `ugreen-diskiomon` service: `modprobe: ERROR: could not insert 'led_ugreen': Exec format error`

I took a guess and imagined that it may need to be reinstalled and possibly also rebuilt for the updated kernel version I was running, so I decided to try building it myself.

Unfortunately, it failed as follows:

```
/tmp/ugreen_leds_controller/kmod$ sudo make
make -C /lib/modules/6.1.0-26-amd64/build M= modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-26-amd64'
  SYNC    include/config/auto.conf.cmd
  GEN     Makefile
/usr/src/linux-headers-6.1.0-26-common/scripts/Makefile.build:44: /usr/src/linux-headers-6.1.0-26-common/scripts/basic/Makefile: No such file or directory
make[3]: *** No rule to make target '/usr/src/linux-headers-6.1.0-26-common/scripts/basic/Makefile'.  Stop.
make[2]: *** [/usr/src/linux-headers-6.1.0-26-common/Makefile:650: scripts_basic] Error 2
/usr/src/linux-headers-6.1.0-26-common/Makefile:792: include/config/auto.conf.cmd: No such file or directory
make[1]: *** [/usr/src/linux-headers-6.1.0-26-common/Makefile:815: include/config/auto.conf.cmd] Error 2
make[1]: *** [include/config/auto.conf.cmd] Deleting file 'include/generated/rustc_cfg'
make[1]: *** [include/config/auto.conf.cmd] Deleting file 'include/generated/autoconf.h'
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-26-amd64'
make: *** [Makefile:11: all] Error 2
```

Frankly, this failure was beyond me at a glance, so I decided to just try reinstalling the `.deb` from the latest release:

```
~$ sudo dpkg -i led-ugreen-dkms_0.1_amd64.deb
Selecting previously unselected package led-ugreen-dkms.
(Reading database ... 87928 files and directories currently installed.)
Preparing to unpack led-ugreen-dkms_0.1_amd64.deb ...
Unpacking led-ugreen-dkms (0.1) ...
Setting up led-ugreen-dkms (0.1) ...
Creating symlink /var/lib/dkms/led-ugreen/0.1/source -> /usr/src/led-ugreen-0.1
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.1.0-26-amd64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-26-amd64 (x86_64)
Consult /var/lib/dkms/led-ugreen/0.1/build/make.log for more information.
```

And the log that dpkg/dkms suggests consulting:

```
~$ cat /var/lib/dkms/led-ugreen/0.1/build/make.log
DKMS make.log for led-ugreen-0.1 for kernel 6.1.0-26-amd64 (x86_64)
Wed 16 Oct 19:06:48 BST 2024
make -C /lib/modules/6.1.0-26-amd64/build M=/var/lib/dkms/led-ugreen/0.1/build modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-26-amd64'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [/usr/src/linux-headers-6.1.0-26-common/Makefile:823: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-26-amd64'
make: *** [Makefile:5: all] Error 2
```

I'm not sure if this is a problem with the project or me, but either way, any advice would be appreciated! 🙂

miskcoo, 2024-10-17:
There are some fixes recenetly. Can you try the new packages [here](https://github.com/miskcoo/ugreen_leds_controller/releases/tag/v0.2-debian12)?

The `led-ugreen` was splitted into two packages, so you may need to remove it first before install the new packages.

miskcoo, 2024-10-18:
> Thanks for the new release! Cool to see utils packaged as a deb too.
> 
> Unfortunately, the dkms install fails in exactly the same way.

How about reinstall the linux-headers?

```
apt install --reinstall linux-headers-6.1.0-26-amd64
```

tigattack, 2024-10-22:
This fixed it, thank you ❤️

closed 2024-10-22 by tigattack