miskcoo/ugreen_leds_controller #42: DKMS fails in Proxmox 8.3.0 (Linux 6.8.12-4)

fishbonbons, 2024-11-27:
The failure thrown out: "Binary /lib/modules/6.8.12-4-pve/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.8.12-4-pve cannot be found at /lib/modules/6.8.12-4-pve/build or /lib/modules/6.8.12-4-pve/source. Please install the linux-headers-6.8.12-4-pve package or use the --kernelsourcedir option to tell DKMS where it's located." 

Describing myself as a noob in linux and scripting, i tried several methods given in the README. I stuck in here - is there a workaround for that?

In the Proxmox VE 8.2 Roadmap is something written about DKMS issues - perhaps related to this issue?!

Thanks for this great piece of code. 
best Sebastian

miskcoo, 2024-11-27:
seems you should install the package mentioned in the error mesaage.

fishbonbons, 2024-11-27:
Thanks for prompt reply. 
As i understand correctly i already have installed the latest packages?! 
uname -r prompt "6.8.12-4-pve"  
Based on an intsallation of Proxmox VE 8.3.0 with Linux 6.8.12-4-pve (2024-11-06T15:04Z)
Furthermore i dont want to brake my installation with doing some stuff close to the kernel…

miskcoo, 2024-11-27:
it is linux-headers-6.8.12-4-pve, not linux-6.8.12-4-pve. This package is
necessary for compiling modules.

If you dont want to use the driver, you can try the cli tool.

miskcoo, 2024-11-27:
besides, if you want to use the scripts for monitoring the disk status,
using pve as a host but running a nas os in virtual machine isn't a good
choice. we have only limited support for this case now.

fishbonbons, 2024-11-27:
Thank you for the clarification. Indeed i run Proxmox as a host on a DXP 4800 Plus with a OMV in a VM. 
All HDD are passed through to the VM, but not the controller. 
I thought i can read s.m.a.r.t. etc. from the HW / devices within the host?!

Alternative, the controller could be handed to the VM, so the VM could act natively with the HDD#s.
Would that be a way it works with ugreen_led_controller?

i tried the cli tools and it worked, but for my understanding i need the kernel modules loaded for a functioning led display?!

miskcoo, 2024-11-28:
i know it wont work if you passthrough the controller, but may  work if
only passthrough the disks.

and yes, the scripts need the kernel module.

fishbonbons, 2024-11-28:
after install the header package it works as intended.

closed 2024-11-28 by fishbonbons