miskcoo/ugreen_leds_controller #43: Power led and nic activity

rybackisback, 2024-12-01:
Ive managed to get it to run. I notice that my power led just keeps blinking. Where do I make changes as per the readme file for power led? is that in the ugreen-led.conf? 

The nic activity led blink script stops working once the NAS restarted or rebooted.  how do you fix this?

Also my DXP 8800 Plus has 2 nics, how do i activate both of them? i repeated the netdevmon command but the led only seems to be picking up activity for one nic.

_Originally posted by @rybackisback in https://github.com/miskcoo/ugreen_leds_controller/issues/13#issuecomment-2509523650_

miskcoo, 2024-12-03:
For the power led, you can add the following code to the second line (below `#!/usr/bin/bash`) of `ugreen-diskiomon` 
```
echo none > /sys/class/leds/power/blink_type
```

`ugreen-netdevmon` cannot show the activities of two NICs now. It depends on the kernel module `ledtrig-netdev` but it only supports linking an led to one NIC.

rybackisback, 2024-12-04:
> For the power led, you can add the following code to the second line (below `#!/usr/bin/bash`) of `ugreen-diskiomon`
>
> ```
> echo none > /sys/class/leds/power/blink_type
> ```
>
> `ugreen-netdevmon` cannot [...]

Thanks @miskcoo that has stopped the power led from blinking!

miskcoo, 2024-12-15:
Seems there are some related contents in #13 resolving your issue.