miskcoo/ugreen_leds_controller #45: ugreen-check-standby not working

0x556c79, 2024-12-01:
The, "ugreen-check-standby" program is not working properly for my DXP8800 Plus.
The color of the LEDs always changes immediately even though the hard disks are active and definitely not in standby.

The HHD Model is WD120EDAZ. But i also had some WD Green 2TB and Seagate 2TB NAS Drive in there and it was the same for them.

How is it determined whether the HDDs are in standby mode? Is there any way I can see this on the shell itself?
Using hdparm? It shows me "unknown" as the status.

```
# hdparm -C /dev/sda

/dev/sda:
SG_IO: bad/missing sense data, sb[]:  f0 00 01 00 50 00 ff 0a 00 00 00 00 00 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive state is:  unknown
```
```
# smartctl -i /dev/sda
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.6.44-production+truenas] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Western Digital Ultrastar (He10/12)
Device Model:     WDC WD120EDAZ-11F3RA0
LU WWN Device Id: 5 000cca 278deccdc
Firmware Version: 81.00A81
User Capacity:    12,000,138,625,024 bytes [12.0 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5400 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database 7.3/5528
ATA Version is:   ACS-2, ATA8-ACS T13/1699-D revision 4
SATA Version is:  SATA 3.2, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Sun Dec  1 17:24:26 2024 CET
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```

_Originally posted by @0x556c79 in https://github.com/miskcoo/ugreen_leds_controller/issues/13#issuecomment-2509542486_

miskcoo, 2024-12-15:
Sorry for the late response. I am quite busy recently.

The program now uses `ioctl` to check the standby status. Maybe it is not suitable for your model since `hdparm` also gives an unknown state about the standby status, but I am not sure.

Alternatively, you can try to run the following command and check the return value of `smartctl` to see whether the standby state can be correctly identified. If it is `1` then the disk is in the standby mode.

```
# smartctl -n standby,1 -H /dev/sda
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.6.59-1-lts] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(1)
```

If it works, I can try to figure out how `smartctl` check this state (when time available).

miskcoo, 2024-12-16:
Here is an minimal example of the `ugreen-check-standby`:

```c++
#include <iostream>
#include <string>
#include <optional>
#include <cstdint>

#include <fcntl.h>
#include <linux/hdreg.h>
#include <sys/ioctl.h>
#include <unistd.h>

std::optional<bool> is_standby_mode(const std::string &device) {

    int fd = open(device.c_str(), O_RDONLY | O_NONBLOCK);
    if (fd == -1) {
        std::cerr << "Failed to open device: " << device << std::endl;
        return std::nullopt;
    }

    unsigned char args[4] = { WIN_CHECKPOWERMODE1, 0, 0, 0 };

    if (ioctl(fd, HDIO_DRIVE_CMD, args) == -1) {
        std::cerr << "ioctl failed in checking power mode of " << device << std::endl;
        close(fd);
        return std::nullopt;
    }

    std::cout << "Returned: " << std::hex 
        << (uint32_t)args[0] << " " 
        << (uint32_t)args[1] << " " 
        << (uint32_t)args[2] << " " 
        << (uint32_t)args[3] << std::endl;

    close(fd);

    return args[2] == 0x00;
}

int main(int argc, char *argv[]) {

    is_standby_mode(argv[1]);

    return 0;
}
```

You can compile it and run to see its outputs.

```bash
g++ <the code>.cpp -o check -O2
./check /dev/sda
```

0x556c79, 2024-12-16:
> Here is an example of the `ugreen-check-standby`:
> 
> ```c++
> #include <iostream>
> #include <string>
> #include <optional>
> #include <cstdint>
> 
> #include <fcntl.h>
> #include <linux/hdreg.h>
> #include <sys/ioctl.h>
> #include <unistd.h>
> 
> std::optional<bool> is_standby_mode(const std::string &device) {
> 
>     int fd = open(device.c_str(), O_RDONLY | O_NONBLOCK);
>     if (fd == -1) {
>         std::cerr << "Failed to open device: " << device << std::endl;
>         return std::nullopt;
>     }
> 
>     unsigned char args[4] = { WIN_CHECKPOWERMODE1, 0, 0, 0 };
> 
>     if (ioctl(fd, HDIO_DRIVE_CMD, args) == -1) {
>         std::cerr << "ioctl failed in checking power mode of " << device << std::endl;
>         close(fd);
>         return std::nullopt;
>     }
> 
>     std::cout << "Returned: " << std::hex 
>         << (uint32_t)args[0] << " " 
>         << (uint32_t)args[1] << " " 
>         << (uint32_t)args[2] << " " 
>         << (uint32_t)args[3] << std::endl;
> 
>     close(fd);
> 
>     return args[2] == 0x00;
> }
> 
> int main(int argc, char *argv[]) {
> 
>     is_standby_mode(argv[1]);
> 
>     return 0;
> }
> ```
> 
> You can compile it and run to see its outputs, and change `WIN_CHECKPOWERMODE1` in the above code to `WIN_CHECKPOWERMODE2` and try again.
> 
> ```shell
> g++ <the code>.cpp -o check -O2
> ./check /dev/sda
> ```

Here is the result.
I have compiled 2 versions, check1 = WIN_CHECKPOWERMODE1 and check2 = WIN_CHECKPOWERMODE2

# ./check1 /dev/sdb
Returned: e5 0 0 0

# ./check2 /dev/sdb
Returned: 98 0 0 0

What do the values mean?

0x556c79, 2024-12-16:
> echo $?

Yes, it is 0