miskcoo/ugreen_leds_controller #48: Proxmox `netdevmon` does not work after boot but after a service restart it does

docgalaxyblock, 2025-01-03:
Hello folks,
first: awesome work on this 👍

I have a DXP8800Plus and just installed Proxmox today on it (`proxmox-headers-6.8.12-5-pve`).
The led controller install went smoothly but on a system restart my netdevmon does not sucessfully starts.
If I manually restart it some seconds later it starts sucessfully.

<details><summary>My <code>journalctl -u ugreen-netdevmon@enp86s0.service</code> log</summary>

```
-- Boot 8c603897bebc4e41986c9a843b2a6717 --
Jan 03 12:28:32 DXP8800 systemd[1]: Starting ugreen-netdevmon@enp86s0.service - UGREEN LEDs daemon for monitoring netio (of enp86s0) and blinking corresponding LEDs...
Jan 03 12:28:32 DXP8800 ugreen-probe-leds[650]: led_ugreen             24576  0
Jan 03 12:28:32 DXP8800 ugreen-probe-leds[631]: Found I2C device /dev/i2c-0
Jan 03 12:28:32 DXP8800 systemd[1]: Started ugreen-netdevmon@enp86s0.service - UGREEN LEDs daemon for monitoring netio (of enp86s0) and blinking corresponding LEDs.
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 38: /sys/class/leds/netdev/trigger: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 39: /sys/class/leds/netdev/device_name: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 40: /sys/class/leds/netdev/link: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 41: /sys/class/leds/netdev/tx: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 42: /sys/class/leds/netdev/rx: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 43: /sys/class/leds/netdev/interval: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 44: /sys/class/leds/netdev/color: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 45: /sys/class/leds/netdev/brightness: No such file or directory
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[675]: cat: /sys/class/net/enp86s0/speed: Invalid argument
Jan 03 12:28:32 DXP8800 ugreen-netdevmon[666]: /usr/bin/ugreen-netdevmon: line 59: /sys/class/leds/netdev/color: No such file or directory
Jan 03 12:30:44 DXP8800 systemd[1]: Stopping ugreen-netdevmon@enp86s0.service - UGREEN LEDs daemon for monitoring netio (of enp86s0) and blinking corresponding LEDs...
Jan 03 12:30:44 DXP8800 ugreen-netdevmon[666]: Terminated
Jan 03 12:30:44 DXP8800 systemd[1]: ugreen-netdevmon@enp86s0.service: Deactivated successfully.
Jan 03 12:30:44 DXP8800 systemd[1]: Stopped ugreen-netdevmon@enp86s0.service - UGREEN LEDs daemon for monitoring netio (of enp86s0) and blinking corresponding LEDs.
Jan 03 12:30:44 DXP8800 systemd[1]: Starting ugreen-netdevmon@enp86s0.service - UGREEN LEDs daemon for monitoring netio (of enp86s0) and blinking corresponding LEDs...
Jan 03 12:30:44 DXP8800 ugreen-probe-leds[1650]: led_ugreen             24576  0
Jan 03 12:30:44 DXP8800 ugreen-probe-leds[1644]: Found I2C device /dev/i2c-0
Jan 03 12:30:44 DXP8800 systemd[1]: Started ugreen-netdevmon@enp86s0.service - UGREEN LEDs daemon for monitoring netio (of enp86s0) and blinking corresponding LEDs.
```
</details>

<details><summary>My installation steps</summary>

```
apt install git build-essential proxmox-default-headers i2c-tools
git clone https://github.com/miskcoo/ugreen_leds_controller
cd ugreen_leds_controller/
wget https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.2/led-ugreen-dkms_0.2_amd64.deb
apt install -f ~/led-ugreen-dkms_0.2_amd64.deb 
cat > /etc/modules-load.d/ugreen-led.conf << EOF
i2c-dev
led-ugreen
ledtrig-oneshot
ledtrig-netdev
EOF
scripts=(ugreen-diskiomon ugreen-netdevmon ugreen-probe-leds); for f in ${scripts[@]}; do chmod +x "scripts/$f"; cp "scripts/$f" /usr/bin; done
cp scripts/ugreen-leds.conf /etc/ugreen-leds.conf
cp scripts/*.service /etc/systemd/system/
cd scripts/
g++ -std=c++17 -O2 blink-disk.cpp -o ugreen-blink-disk
cp ugreen-blink-disk /usr/bin
g++ -std=c++17 -O2 check-standby.cpp -o ugreen-check-standby
cp ugreen-check-standby /usr/bin
nano /etc/ugreen-leds.conf
systemctl enable --now ugreen-diskiomon
systemctl enable --now ugreen-netdevmon@enp86s0
reboot
systemctl status ugreen-diskiomon.service ugreen-netdevmon@enp86s0.service
systemctl restart ugreen-diskiomon.service ugreen-netdevmon@enp86s0.service
``` 
</details>

I see systemd dependencies were included in #23 so it should work right?
Or did I missed something during installation?

PS: The `diskiomon` service works directly after boot.

---

Greetings from 🇩🇪 👋

miskcoo, 2025-01-03:
If restarting `ugreen-netdevmon@enp86s0.service` would make it work, maybe inserting a `sleep 2s` below [this line](https://github.com/miskcoo/ugreen_leds_controller/blob/master/scripts/ugreen-probe-leds#L14C1-L14C80) will resolve this issue.

jrcheshire, 2025-01-03:
I've been encountering a very similar issue with my DXP4800 Plus on Debian 12. Either the network monitor or disk I/O monitor LEDs would work, but not both, and it seemed to be random which one worked each time. Errors of the form
`ugreen-probe-leds[552]: /usr/bin/ugreen-probe-leds: line 14: echo: write error: Device or resource busy`, and similarly restarting the service manually could recover whichever one didn't work.

I played around a bit with adding `sleep` statements per the above suggestion, initially a 2-second sleep below line 14 of `ugreen-probe-leds` as suggested, and played around with also adding a sleep before this line and making the sleep duration longer (went up to 5 seconds max). This didn't seem to have the desired effect, but what _did_ was adding 3-second sleep statements at the top of both the `ugreen-netdevmon` and `ugreen-diskiomon` scripts.

docgalaxyblock, 2025-01-03:
Year I experimented a bit too *(I have setup a fresh pve on a test ssh on my dxp8800).*
If I disable netdevmon and diskiomon, restart it and then manually run ` /usr/bin/ugreen-probe-leds` I get no error.
I am currently guessing both services get started nearly in parallel and therefore the probe-leds ExecStartPre gets executed nearly in parallel too and the "second" instanc of probe-leds gts the Ressource bussy error.

I will try to move the probe-leds into its own service and depend the original services on it.
Let's see if I get this changed and if it works.

docgalaxyblock, 2025-01-04:
I was able to catch two different errors with the current services:

<details><summary>A) <code>ugreen-probe-leds: line 14: echo: write error: File exists</code></summary>

```
root@DXP8800:~# systemctl status -o short-iso-precise ugreen-{probe-led,diskiomon,netdevmon@vmbr0}
Unit ugreen-probe-led.service could not be found.
× ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-diskiomon.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2025-01-04 00:10:21 CET; 1min 6s ago
    Process: 678 ExecStartPre=/usr/bin/ugreen-probe-leds (code=exited, status=1/FAILURE)
        CPU: 16ms

2025-01-04T00:10:21.795178+0100 DXP8800 systemd[1]: Starting ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs...
2025-01-04T00:10:21.807703+0100 DXP8800 ugreen-probe-leds[702]: led_ugreen             24576  0
2025-01-04T00:10:21.811473+0100 DXP8800 ugreen-probe-leds[678]: Found I2C device /dev/i2c-0
2025-01-04T00:10:21.812329+0100 DXP8800 ugreen-probe-leds[678]: /usr/bin/ugreen-probe-leds: line 14: echo: write error: File exists
2025-01-04T00:10:21.811780+0100 DXP8800 systemd[1]: ugreen-diskiomon.service: Control process exited, code=exited, status=1/FAILURE
2025-01-04T00:10:21.811905+0100 DXP8800 systemd[1]: ugreen-diskiomon.service: Failed with result 'exit-code'.
2025-01-04T00:10:21.812060+0100 DXP8800 systemd[1]: Failed to start ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs.

○ ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-netdevmon@.service; enabled; preset: enabled)
     Active: inactive (dead) since Sat 2025-01-04 00:10:22 CET; 1min 6s ago
   Duration: 54ms
    Process: 679 ExecStartPre=/usr/bin/ugreen-probe-leds (code=exited, status=0/SUCCESS)
    Process: 782 ExecStart=/usr/bin/ugreen-netdevmon vmbr0 (code=exited, status=0/SUCCESS)
   Main PID: 782 (code=exited, status=0/SUCCESS)
        CPU: 28ms

2025-01-04T00:10:21.795612+0100 DXP8800 systemd[1]: Starting ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs...
2025-01-04T00:10:21.808100+0100 DXP8800 ugreen-probe-leds[700]: led_ugreen             24576  0
2025-01-04T00:10:21.811209+0100 DXP8800 ugreen-probe-leds[679]: Found I2C device /dev/i2c-0
2025-01-04T00:10:22.542812+0100 DXP8800 systemd[1]: Started ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs.
2025-01-04T00:10:22.597234+0100 DXP8800 systemd[1]: ugreen-netdevmon@vmbr0.service: Deactivated successfully.
```

</details>

<details><summary>B) <code>ugreen-netdevmon: line 38: /sys/class/leds/netdev/trigger: No such file or directory ...</code></summary>

```
root@DXP8800:~# systemctl status -o short-iso-precise ugreen-{probe-led,diskiomon,netdevmon@vmbr0}
Unit ugreen-probe-led.service could not be found.
● ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-diskiomon.service; enabled; preset: enabled)
     Active: active (running) since Fri 2025-01-03 23:53:08 CET; 12min ago
    Process: 684 ExecStartPre=/usr/bin/ugreen-probe-leds (code=exited, status=0/SUCCESS)
   Main PID: 785 (ugreen-diskiomo)
      Tasks: 6 (limit: 47734)
     Memory: 3.5M
        CPU: 19.286s
     CGroup: /system.slice/ugreen-diskiomon.service
             ├─  785 /usr/bin/bash /usr/bin/ugreen-diskiomon
             ├─  962 /usr/bin/bash /usr/bin/ugreen-diskiomon
             ├─  963 /usr/bin/bash /usr/bin/ugreen-diskiomon
             ├─24383 sleep 360s
             ├─25532 sleep 5s
             └─25589 sleep 0.1s

2025-01-03T23:53:07.654387+0100 DXP8800 systemd[1]: Starting ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs...
2025-01-03T23:53:07.666460+0100 DXP8800 ugreen-probe-leds[705]: led_ugreen             24576  0
2025-01-03T23:53:07.670388+0100 DXP8800 ugreen-probe-leds[684]: Found I2C device /dev/i2c-0
2025-01-03T23:53:08.341765+0100 DXP8800 systemd[1]: Started ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs.
2025-01-03T23:53:08.347506+0100 DXP8800 ugreen-diskiomon[785]: Found UGREEN DXP8800 series
2025-01-03T23:53:08.351322+0100 DXP8800 ugreen-diskiomon[785]: Enumerating disks based on ata...
2025-01-03T23:53:08.353020+0100 DXP8800 ugreen-diskiomon[785]: ata ata2 >> sda

○ ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-netdevmon@.service; enabled; preset: enabled)
[... 2 lines cut ...]
    Process: 685 ExecStartPre=/usr/bin/ugreen-probe-leds (code=exited, status=0/SUCCESS)
    Process: 721 ExecStart=/usr/bin/ugreen-netdevmon vmbr0 (code=exited, status=0/SUCCESS)
   Main PID: 721 (code=exited, status=0/SUCCESS)
        CPU: 23ms

2025-01-03T23:53:07.654834+0100 DXP8800 systemd[1]: Starting ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs...
2025-01-03T23:53:07.665316+0100 DXP8800 ugreen-probe-leds[703]: led_ugreen             24576  0
[... cut ...]
2025-01-03T23:53:07.679941+0100 DXP8800 ugreen-netdevmon[721]: /usr/bin/ugreen-netdevmon: line 44: /sys/class/leds/netdev/color: No such file or directory
2025-01-03T23:53:07.679941+0100 DXP8800 ugreen-netdevmon[721]: /usr/bin/ugreen-netdevmon: line 45: /sys/class/leds/netdev/brightness: No such file or directory
2025-01-03T23:53:07.680859+0100 DXP8800 systemd[1]: ugreen-netdevmon@vmbr0.service: Deactivated successfully.
```

</details>

I am guessing case A) happens cause of the parallelism of the `probe-leds` `ExecStartPre`

In case B) I am not sure...

miskcoo, 2025-01-04:
I think the second issue arises because there is no additional waiting time for the driver's initialization. How about adding `sleep 2s`, as mentioned earlier? Or adding another `ExecStartPre=sleep 2s` after the `probe-leds`.

docgalaxyblock, 2025-01-04:
I got it - I moved the probe-leds into its own service of the type oneshot and depend the diskmon & diskiomon thanks to this post since I did not looked into systemd yet:
https://trstringer.com/simple-vs-oneshot-systemd-service

<details><summary>Now the dependency is correct and it correctly started multiple times in my case</summary>

```bash
root@DXP8800:~# systemctl status -o short-iso-precise systemd-modules-load ugreen-{probe-led,diskiomon,netdevmon@vmbr0}

● systemd-modules-load.service - Load Kernel Modules
     Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static)
     Active: active (exited) since Sat 2025-01-04 16:49:03 CET; 25s ago
       Docs: man:systemd-modules-load.service(8)
             man:modules-load.d(5)
    Process: 403 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=0/SUCCESS)
   Main PID: 403 (code=exited, status=0/SUCCESS)
        CPU: 1.365s

2025-01-04T16:49:01.945891+0100 DXP8800 systemd-modules-load[403]: Inserted module 'vhost_net'
2025-01-04T16:49:01.946552+0100 DXP8800 systemd-modules-load[403]: Module 'i2c_dev' is built in
2025-01-04T16:49:01.948807+0100 DXP8800 systemd-modules-load[403]: Inserted module 'led_ugreen'
2025-01-04T16:49:01.952653+0100 DXP8800 systemd-modules-load[403]: Inserted module 'ledtrig_oneshot'
2025-01-04T16:49:01.954038+0100 DXP8800 systemd-modules-load[403]: Inserted module 'ledtrig_netdev'
2025-01-04T16:49:03.392699+0100 DXP8800 systemd-modules-load[403]: Inserted module 'zfs'
2025-01-04T16:49:03.393459+0100 DXP8800 systemd[1]: Finished systemd-modules-load.service - Load Kernel Modules.
Notice: journal has been rotated since unit was started, output may be incomplete.

● ugreen-probe-led.service - UGREEN LEDs daemon for probing the hardare LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-probe-led.service; disabled; preset: enabled)
     Active: active (exited) since Sat 2025-01-04 16:49:04 CET; 24s ago
    Process: 679 ExecStart=/usr/bin/ugreen-probe-leds (code=exited, status=0/SUCCESS)
   Main PID: 679 (code=exited, status=0/SUCCESS)
        CPU: 20ms

2025-01-04T16:49:03.675651+0100 DXP8800 systemd[1]: Starting ugreen-probe-led.service - UGREEN LEDs daemon for probing the hardare LEDs...
2025-01-04T16:49:03.689080+0100 DXP8800 ugreen-probe-leds[690]: led_ugreen             24576  0
2025-01-04T16:49:03.696522+0100 DXP8800 ugreen-probe-leds[679]: Found I2C device /dev/i2c-0
2025-01-04T16:49:04.473648+0100 DXP8800 systemd[1]: Finished ugreen-probe-led.service - UGREEN LEDs daemon for probing the hardare LEDs.

● ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-diskiomon.service; enabled; preset: enabled)
     Active: active (running) since Sat 2025-01-04 16:49:04 CET; 24s ago
   Main PID: 774 (ugreen-diskiomo)
      Tasks: 6 (limit: 47734)
     Memory: 3.1M
        CPU: 649ms
     CGroup: /system.slice/ugreen-diskiomon.service
             ├─ 774 /usr/bin/bash /usr/bin/ugreen-diskiomon
             ├─1009 /usr/bin/bash /usr/bin/ugreen-diskiomon
             ├─1010 /usr/bin/bash /usr/bin/ugreen-diskiomon
             ├─1018 sleep 360s
             ├─1747 sleep 5s
             └─1901 sleep 0.1s

2025-01-04T16:49:04.489363+0100 DXP8800 systemd[1]: Started ugreen-diskiomon.service - UGREEN LEDs daemon for monitoring diskio and blinking corresponding LEDs.
2025-01-04T16:49:04.495281+0100 DXP8800 ugreen-diskiomon[774]: Found UGREEN DXP8800 series
2025-01-04T16:49:04.498896+0100 DXP8800 ugreen-diskiomon[774]: Enumerating disks based on ata...
2025-01-04T16:49:04.500924+0100 DXP8800 ugreen-diskiomon[774]: ata ata2 >> sda

○ ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs
     Loaded: loaded (/etc/systemd/system/ugreen-netdevmon@.service; enabled; preset: enabled)
     Active: inactive (dead) since Sat 2025-01-04 16:49:04 CET; 24s ago
   Duration: 54ms
    Process: 775 ExecStart=/usr/bin/ugreen-netdevmon vmbr0 (code=exited, status=0/SUCCESS)
   Main PID: 775 (code=exited, status=0/SUCCESS)
        CPU: 7ms

2025-01-04T16:49:04.490335+0100 DXP8800 systemd[1]: Started ugreen-netdevmon@vmbr0.service - UGREEN LEDs daemon for monitoring netio (of vmbr0) and blinking corresponding LEDs.
2025-01-04T16:49:04.544343+0100 DXP8800 systemd[1]: ugreen-netdevmon@vmbr0.service: Deactivated successfully.
```

</details>

Since I do not know systemd yet and reading the mentioned post I think that the `netdevmon@.servce` could be switched to oneshot with RemainAfterExit too.

What are your opinions?

docgalaxyblock, 2025-01-04:
Hey @jrcheshire @miskcoo and maybe @KrzysztofHajdamowicz Would you like to test my changes before I open a PR?

https://github.com/docgalaxyblock/ugreen_leds_controller

miskcoo, 2025-01-05:
> Would you like to test my changes before I open a PR?

I can test that recently.

> Since I do not know systemd yet and reading the mentioned post I think that the netdevmon@.servce could be switched to oneshot with RemainAfterExit too.

I will also check that.

miskcoo, 2025-01-10:
Hi @docgalaxyblock, I have tested the changes in the repo, and they work well.

For the `netdevmon@.service`, it behaves similar to `diskiomon.service` since there is a loop for checking the network connectivity, so I think it should be a service of type simple.

docgalaxyblock, 2025-01-11:
> For the netdevmon@.service, it behaves similar to diskiomon.service since there is a loop for checking the network connectivity, so I think it should be a service of type simple.

@miskcoo It depends if you have enabled CHECK_LINK or not. If not it is likely a "oneshot" type of service.

miskcoo, 2025-01-11:
> > For the [netdevmon@.service](mailto:netdevmon@.service), it behaves similar to diskiomon.service since there is a loop for checking the network connectivity, so I think it should be a service of type simple.
> 
> @miskcoo It depends if you have enabled CHECK_LINK or not. If not it is likely a "oneshot" type of service.

By looking the link about systemd, the oneshot type with remainafterexit leaves the service to the activating state with check_link enabled. It appears that the requirements for these two choices are slightly conflict?

docgalaxyblock, 2025-01-12:
> It appears that the requirements for these two choices are slightly conflict?
Year I interpret it like that too.

docgalaxyblock, 2025-01-18:
I understand it like that too

PR #51 (2025-01-11): fix systemd services
Fixes #48