miskcoo/ugreen_leds_controller #52: Provide CLI

S2ciOnur, 2025-01-28:
Hey, is it possible that you provide the .cli?
It is not that simple to create a own using make for a noob :/

miskcoo, 2025-01-28:
I think it is already available here:

https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.1-debian12/ugreen_leds_cli

S2ciOnur, 2025-01-29:
> I think it is already available here:
> 
> https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.1-debian12/ugreen_leds_cli

isn't this the "old" version and there is a newer one?

miskcoo, 2025-01-30:
The cli tool is not updated recently, and that version is usable. I also updated the latest version and you can try it.

https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.2/ugreen_leds_cli

closed 2025-01-31 by S2ciOnur