miskcoo/ugreen_leds_controller #53: Mass of led_ugreen messages in console

Laurenzis, 2025-03-02:
Is there any possibilty to prevent led_ugreen messages written into console? Due to a raid-rebuild I have about >10000 messages per hour...

miskcoo, 2025-03-03:
Hi, the version you referred has the verbose parameter disabled by default. Using it directly may also hide the message. 

You can check and modify it dynamically:

```bash
cat /sys/module/led_ugreen/parameters/verbose
# disable the message
# echo N > /sys/module/led_ugreen/parameters/verbose
```

miskcoo, 2025-03-03:
For the failure to turn the LED on or off, I am not sure what the issue is. Can you provide your NAS model?

Laurenzis, 2025-03-03:
I'm sorry, it's a ugreen DXP6800pro, running with latest version of OMV.

Laurenzis, 2025-03-03:
> You can check and modify it dynamically:
> cat /sys/module/led_ugreen/parameters/verbose
Actually, there is no folder "parameters" in /sys/module/led_ugreen.

Only the following exists:
drivers
holders
notes
sections

miskcoo, 2025-03-20:
I realized that I forgot to update the release for Debian. If you installed the packages (v0.2) from the release page, you can upgrade to v0.3 then the parameters folder may be available.