miskcoo/ugreen_leds_controller #55: Compatibility with TrueNAS 25.04

JonRowe, 2025-04-17:
👋 Thanks for making this library, I'm trying to use it on a 6800 Pro with TrueNAS 25.04.

I've been able to build (and use) the cli but attempts to build the kernel module results are running into a sudo bug (even when run as root):

```
root@nas[~/ugreen_leds_controller/kmod]# make
make -C /lib/modules/6.12.15-production\+truenas/build M=/root/ugreen_leds_controller/kmod modules
make[1]: Entering directory '/usr/src/linux-headers-truenas-production-amd64'
sudo: argv[2] mismatch, expected "set -e;  echo '  CC [M]  /root/ugreen_leds_controller/kmod/led-ugreen.o'; ...
```

miskcoo, 2025-04-17:
How about using the prebuilt version [here](https://github.com/miskcoo/ugreen_leds_controller/tree/gh-actions/build-scripts/truenas/build/TrueNAS-SCALE-Fangtooth/25.04.0)?

rybackisback, 2025-04-17:
@miskcoo when i try and follow the instructions with TrueNAS 25.04 fangtooth using the prebuilt kernel module i get the following error:
cp: cannot create regular file '/lib/modules/6.6.44-production+truenas/extra': No such file or directory

deepybee, 2025-04-22:
@rybackisback Fangtooth 25.04.0 [uses kernel 6.12.15](https://www.truenas.com/docs/scale/25.04/gettingstarted/scalereleasenotes/#component-versions), not 6.6.44. Update the version number in the path and it should work as expected.

```
uname -a
Linux myhostname 6.12.15-production+truenas #1 SMP PREEMPT_DYNAMIC Mon Apr 14 14:41:12 UTC 2025 x86_64 GNU/Linux
```

closed 2025-07-04 by miskcoo