miskcoo/ugreen_leds_controller #56: Ignore "smartctl -H" return codes for past errors

heldchen, 2025-06-03:
First of all, thanks for the useful tools!

I have some drives in my nas that overheated in the past. this results `smartctl -H` to set the bit 5 ("SMART status check returned "DISK OK" but some attributes have been at or below threshold in the past.") of the return code, otherwise the disk is perfectly fine:

```
# smartctl -H -l error /dev/sda
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.11.0-26-generic] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED
Please note the following marginal Attributes:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
190 Airflow_Temperature_Cel 0x0022   061   042   045    Old_age   Always   In_the_past 39 (5 252 39 28 0)

SMART Error Log Version: 1
No Errors Logged
```

currently, the `scripts/ugreen-diskiomon` treats all non-zero return codes as failures:
```
            /usr/sbin/smartctl -H /dev/${dev} -n standby,1 &> /dev/null
            RET=$?

            if [[ $RET -gt 1 ]]; then
                echo "$COLOR_SMART_FAIL" > /sys/class/leds/$led/color
                echo Disk failure detected on /dev/$dev at $(date +%Y-%m-%d' '%H:%M:%S)
                continue
            fi
```
thus my disks are now marked as failed (red leds). 
it would be nice if non-critical issues could be ignored from the check. the meaning for each bit can be found [here](https://invent.kde.org/system/kpmcore/-/merge_requests/28). as all other bits seem to be critical, a simple xor check against bit 5 would exclude past errors:

```
            /usr/sbin/smartctl -H /dev/${dev} -n standby,1 &> /dev/null
            RET=$?

            # check return code for critical errors (any bit set except bit 5)
            if (( $RET & ~32 )); then
                echo "$COLOR_SMART_FAIL" > /sys/class/leds/$led/color
                echo Disk failure detected on /dev/$dev at $(date +%Y-%m-%d' '%H:%M:%S)
                continue
            fi
```

miskcoo, 2025-06-04:
Thanks for the suggestion, which makes perfect sense.

Would you be interested in submitting a pull request?

heldchen, 2025-06-04:
absolutely! here it is: https://github.com/miskcoo/ugreen_leds_controller/pull/57

closed 2025-06-04 by miskcoo