miskcoo/ugreen_leds_controller #58: Disks that are idle are marked as failed

heldchen, 2025-06-05:
As my old Seagate 8tb SMR drives are very noisy I am using [hd-idle](https://github.com/adelolmo/hd-idle) to put them to sleep after 30min. when this happens, ugreen-diskiomon marks them as failed:
```
Jun 05 16:56:20 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sdh at 2025-06-05 16:56:20
Jun 05 16:56:20 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sdf at 2025-06-05 16:56:20
Jun 05 16:56:20 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sdg at 2025-06-05 16:56:20
Jun 05 16:56:20 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sdd at 2025-06-05 16:56:20
Jun 05 16:56:20 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sde at 2025-06-05 16:56:20
Jun 05 16:56:21 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sdb at 2025-06-05 16:56:21
Jun 05 16:56:21 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sdc at 2025-06-05 16:56:21
Jun 05 16:56:21 kevin ugreen-diskiomon[56089]: Disk failure detected on /dev/sda at 2025-06-05 16:56:21
```

even after waking the disks up again, the led [stay red without activity indication](https://github.com/miskcoo/ugreen_leds_controller/blob/276ae34cd15273df7233dfcdb4675713b2b533ea/scripts/ugreen-diskiomon#L271C13-L271C63). 

unfortunately just probing the disks again with smartctl would wake them up and as such defeat the purpose of hd-idle. I don't know linux' handling of drives well enough, but maybe there's a /dev property to check the drive's sleep status?

miskcoo, 2025-06-06:
Can you check the return code of smartctl at [this line](https://github.com/miskcoo/ugreen_leds_controller/blob/276ae34cd15273df7233dfcdb4675713b2b533ea/scripts/ugreen-diskiomon#L277)? I think the parameter `-n standby,1` avoids waking up the disk and yields a return code `1` as below (see #37): 

```
# smartctl -H /dev/sda -n standby,1; echo $?
smartctl 7.5 2025-04-30 r5714 [x86_64-linux-6.12.20-1-lts] (local build)
Copyright (C) 2002-25, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(1)
1
```

Besides, there seems to be a difference between the sleep state (`hdparm -C /dev/sda`) and the standby mode (`hdparm -y /dev/sda`). The above command will wake a disk from the sleep state but not from standby mode, and it appears to me that the standby mode is sufficient to keep the disk quiet.

heldchen, 2025-06-06:
here's the output of a disk in standby:
```
$ sudo smartctl -H /dev/sda -n standby,1; echo $?
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.11.0-26-generic] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(1)
1
```

closed 2025-06-07 by miskcoo