miskcoo/ugreen_leds_controller #64: Release for Trixie

djdanjones, 2025-08-12:
Given latest (9.0) release of proxmox is Trixie, I think we need a new release with new kernel module for that.
Any chance of this release being done?

Ideally don't want to load dev-tools to my PVE environment to compile a kernel module!

djdanjones, 2025-08-12:
Well I didn't have the hardware needing it on pve 8.0 on bookworm ... upgraded to 9.0 and trixie on new kit.    I havn't tired the existing deb so will try that if thats okay on trixie!

mevgmbh, 2025-08-12:
I just installed Debian13 on a DX2800 and added to to my internal LED configuration:
# Front-LED	https://github.com/miskcoo/ugreen_leds_controller
# 		systemd dir -> /etc/systemd/system/ugreen*
#		scripts dir -> /usr/bin/ugreen*
# FL-Install1	https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.3/led-ugreen-dkms_0.3_amd64.deb  -> apt install dkim&et al.
# FL-Install2	https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.3/led-ugreen-utils_0.3_amd64.deb -> dkms install -m led-ugreen -v 0.3 -> /usr/bin/ugreen-probe-leds -> OK
# FL-HD		systemctl enable ugreen-diskiomon.service
# FL-Network	systemctl start ugreen-netdevmon2.service, modify nano /etc/systemd/system/ugreen-netdevmon@.service to with "nano ugreen-netdevmon2.service" to fit to "ip a",enp2s0
#
=> LED's are working well, Thanks!

djdanjones, 2025-08-12:
Thanks -> got it working in end on proxmox 9.0 (ie, released last week).   Needed custom dkms, but wasn't a issue.

Thanks

closed 2025-08-12 by djdanjones