miskcoo/ugreen_leds_controller #66: DXP2800 i2c_smbus_read_i2c_block_data failed with id 0

Leonick, 2025-08-30:
I've installed ich777's Unraid plugin on my DXP2800, sadly it isn't working. ich777 hadn't seen these errors before and suggested I should create an issue here. Any ideas?

`Aug 30 22:24:15 NAS kernel: led_ugreen: initializing
Aug 30 22:24:15 NAS emhttpd: shcmd (5): /usr/local/emhttp/webGui/scripts/update_access
Aug 30 22:24:15 NAS emhttpd: shcmd (6): modprobe zfs
Aug 30 22:24:15 NAS kernel: led_ugreen: i2c probed
Aug 30 22:24:15 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 0, err -110
Aug 30 22:24:15 NAS kernel: zfs: module license 'CDDL' taints kernel.
Aug 30 22:24:15 NAS kernel: Disabling lock debugging due to kernel taint
Aug 30 22:24:15 NAS kernel: zfs: module license taints kernel.
Aug 30 22:24:16 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 0, err -110
Aug 30 22:24:16 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 0, err -110
Aug 30 22:24:16 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 0, err -110
Aug 30 22:24:16 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 0, err -110
Aug 30 22:24:16 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:17 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 1, err -16
Aug 30 22:24:17 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 1, err -110
Aug 30 22:24:17 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 1, err -110
Aug 30 22:24:17 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 1, err -110
Aug 30 22:24:17 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 1, err -110
Aug 30 22:24:17 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:17 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 2, err -16
Aug 30 22:24:17 NAS kernel: ZFS: Loaded module v2.3.1-1, ZFS pool version 5000, ZFS filesystem version 5
Aug 30 22:24:18 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 2, err -110
Aug 30 22:24:18 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 2, err -110
Aug 30 22:24:18 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 2, err -110
Aug 30 22:24:18 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 2, err -110
Aug 30 22:24:18 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:19 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 3, err -16
Aug 30 22:24:19 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 3, err -110
Aug 30 22:24:19 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 3, err -110
Aug 30 22:24:19 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 3, err -110
Aug 30 22:24:19 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 3, err -110
Aug 30 22:24:19 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:19 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 4, err -16
Aug 30 22:24:20 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 4, err -110
Aug 30 22:24:20 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 4, err -110
Aug 30 22:24:20 NAS kernel: tun: Universal TUN/TAP device driver, 1.6
Aug 30 22:24:20 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 4, err -110
Aug 30 22:24:20 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 4, err -110
Aug 30 22:24:20 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:20 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 5, err -16
Aug 30 22:24:21 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 5, err -110
Aug 30 22:24:21 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 5, err -110
Aug 30 22:24:21 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 5, err -110
Aug 30 22:24:21 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 5, err -110
Aug 30 22:24:21 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:21 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 6, err -16
Aug 30 22:24:22 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 6, err -110
Aug 30 22:24:22 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 6, err -110
Aug 30 22:24:22 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 6, err -110
Aug 30 22:24:22 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 6, err -110
Aug 30 22:24:22 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:22 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 7, err -16
Aug 30 22:24:23 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 7, err -110
Aug 30 22:24:23 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 7, err -110
Aug 30 22:24:23 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 7, err -110
Aug 30 22:24:23 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 7, err -110
Aug 30 22:24:23 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:23 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 8, err -16
Aug 30 22:24:24 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 8, err -110
Aug 30 22:24:24 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 8, err -110
Aug 30 22:24:24 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 8, err -110
Aug 30 22:24:24 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 8, err -110
Aug 30 22:24:24 NAS kernel: i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
Aug 30 22:24:24 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 9, err -16
Aug 30 22:24:25 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 9, err -110
Aug 30 22:24:25 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 9, err -110
Aug 30 22:24:25 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 9, err -110
Aug 30 22:24:25 NAS kernel: led_ugreen: ugreen_led_get_state: i2c_smbus_read_i2c_block_data failed with id 9, err -110
Aug 30 22:24:25 NAS kernel: i2c i2c-0: new_device: Instantiated device led-ugreen at 0x3a`

miskcoo, 2025-08-31:
I hadn't seen it before either. It seems the driver cannot communicate with the chip that controls the LEDs in your NAS. Could you try the `i2cdetect -l` and `i2c-detect -y 0` commands from the [readme](https://github.com/miskcoo/ugreen_leds_controller/tree/master?tab=readme-ov-file#preparation) and the `ugreen_leds_cli` tool therein and check whether they give the desired outputs (you may need to disable this plugin first)?

Leonick, 2025-08-31:
```
root@NAS:~# i2cdetect -l
i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-1   i2c             i915 gmbus dpa                          I2C adapter
i2c-2   i2c             i915 gmbus dpb                          I2C adapter
i2c-3   i2c             i915 gmbus dpc                          I2C adapter
i2c-4   i2c             i915 gmbus tc1                          I2C adapter
i2c-5   i2c             i915 gmbus tc2                          I2C adapter
i2c-6   i2c             i915 gmbus tc3                          I2C adapter
i2c-7   i2c             i915 gmbus tc4                          I2C adapter
i2c-8   i2c             i915 gmbus tc5                          I2C adapter
i2c-9   i2c             i915 gmbus tc6                          I2C adapter
root@NAS:~# i2c-detect -y 0
bash: i2c-detect: command not found
root@NAS:~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- 4a -- -- -- -- -- 
50: -- -- 52 -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@NAS:~#
```

Not quite sure how to get the ugreen_leds_cli, the plugin doesn't include it and unraid doesn't have the make command. Only finding ways to install it for older versions of unraid.

miskcoo, 2025-08-31:
You can download it [here](https://github.com/miskcoo/ugreen_leds_controller/releases/download/v0.3/ugreen_leds_cli), but from the output of i2cdetect the address `3a` has been used by some drivers.

If `cat /sys/class/i2c-dev/i2c-0/device/0-003a/name` gives `led-ugreen` then you may need to unload this driver by `rmmod led-ugreen` before using the cli tool.

Leonick, 2025-08-31:
-status gives no output and -off didn't turn them off.
```
root@NAS:~# ./ugreen_leds_cli all -status
root@NAS:~# ./ugreen_leds_cli all -off
root@NAS:~# 
```

miskcoo, 2025-09-02:
Strange. So when `led-ugreen` is removed, is the output of `i2cdetect -y 0` at slot `3A` still `UU`, and what is the output of `cat /sys/class/i2c-dev/i2c-0/device/0-003a/name`?

Leonick, 2025-09-02:
```
root@NAS:~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- 3a -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- 4a -- -- -- -- -- 
50: -- -- 52 -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@NAS:~# cat /sys/class/i2c-dev/i2c-0/device/0-003a/name
led-ugreen
```