miskcoo/ugreen_leds_controller #68: cli version is working on trixie

Luke56784, 2025-09-13:
CLI is working on debian 13.

Tested on DXP4800

Luke56784, 2025-09-13:
I wanted to share my systemd unit file.

It loads the kernel module and then deactivates everything except the power led.

Maybe you can put this in as an example for beginners who just want to get rid of the flashing?

```
[Unit]
Description=Disable UGREEN NAS LEDs on boot
After=multi-user.target
Wants=systemd-modules-load.service

[Service]
Type=oneshot
ExecStart=/sbin/modprobe i2c-dev
ExecStart=PATHTOCLI/ugreen_leds_cli netdev disk1 disk2 disk3 disk4 -off

[Install]
WantedBy=multi-user.target
```

henri98, 2025-11-16:
Can confirm this on the dxp 2800.

```
root@dxp-2800:/usr/bin# ugreen-probe-leds 
i2c_dev                28672  0
led_ugreen             28672  0
Found I2C device /dev/i2c-0
root@dxp-2800:/usr/bin# ugreen_leds_cli all -off 
Err: fail to open the I2C device.
Please check that (1) you have the root permission; 
              and (2) the i2c-dev module is loaded. 
```